ethz-asl/maplab #2: Protobuf Build Issue

goldbattle, 2017-11-29:
Having a issue building this with [catkin tools](https://catkin-tools.readthedocs.io/en/latest/).
It seems that the [protobuf_catkin](https://github.com/ethz-asl/protobuf_catkin) library does not support isolated building?
Error I am getting on ubuntu 16.04 and related [link](https://answers.ros.org/question/248718/catkin_devel_prefix-in-catkin_make-vs-catkin-tools/):
```cmd                                                   
make[2]: /home/xxxx/workspace/catkin_ws_maplab/devel/.private/maplab_common/bin/protoc: Command not found
make[2]: *** [compiled_proto/maplab-common/id.pb.cc] Error 127
make[2]: *** Waiting for unfinished jobs....
make[2]: /home/xxxx/workspace/catkin_ws_maplab/devel/.private/maplab_common/bin/protoc: Command not found
make[2]: *** [compiled_proto/maplab-common/eigen.pb.cc] Error 127
make[1]: *** [CMakeFiles/maplab_common.dir/all] Error 2
make: *** [all] Error 2
```

goldbattle, 2017-11-29:
I am able to build using the merged devel workspace.
```cmd
catkin config --merge-devel
```
Would be nice to have a note in the FAQ as I skipped that setup step entirely and expected just cloning the two repositories to work out of the box.

eggerk, 2017-11-30:
Thanks for your report!

I added a section about this to the [wiki](https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu#protobuf-issue-protoc-command-not-found).

closed 2017-11-30 by eggerk