ethz-asl/maplab #3: Missing Dependencies

goldbattle, 2017-11-29:
Had some trouble doing a fresh install on ubuntu 16.04 needed to clone these two repositories to successfully build all packages from maplab and maplab_dependencies.

* https://github.com/ethz-asl/oomact
* https://github.com/ethz-asl/asl_cmake_modules
* https://github.com/ethz-asl/aslam_optimizer
* https://github.com/ethz-asl/aslam_splines
* https://github.com/ethz-asl/schweizer_messer

Seems that `hand_eye_calibration_batch_estimation` depends on oomact, with oomact depending on those last four links.

* * * 

Also the hand_eye_calibration is at `ce88f43e655841ce609f8a0a5e7ec301d015e03d` while the newest `f75a8e8ffb791292e5fbb127298d8a4ed32677ec` is needed to correct the calibration [header includes](https://github.com/ethz-asl/hand_eye_calibration/commit/8936aff36bda6a3057f014e4543ea97932c11d96). I assume it is because I am using the newest oomact as this was not included in the dependency repo.

mfehr, 2017-11-30:
@goldbattle thanks for the report. 

Do you have any missing dependencies if you build only maplab, i.e. `catkin build maplab`?

Some of our dependencies (e.g. `hand_eye_calibration`) include additional packages which are not needed (e.g. `hand_eye_calibration_batch_estimation`). If you are trying to build these additional packages as well, you will inevitably run into missing dependencies, as our installation instructions only include the packages required for maplab.

This might change in the future, but in the meantime, if you prefer to build the whole workspace, you can add `CATKIN_IGNORE` files to the couple of packages that are not fully supported by our dependencies.

goldbattle, 2017-11-30:
Looks like it needed for a test package.
If this one is not built then it should all be fine.
https://github.com/ethz-asl/maplab/blob/master/test/end-to-end-common/package.xml

mfehr, 2017-11-30:
The hand_eye_calibration **repository** contains multiple packages, not all of which can be built, however, the one we depend on, the `hand_eye_calibration` **package**, has all the dependencies it needs, Therefore if you run `catkin build maplab` it should build just fine. Let me know if this is not the case.

closed 2017-12-06 by goldbattle