ethz-asl/maplab #4: Kalibr Camchain Conversion

goldbattle, 2017-11-29:
When using the [kalibr_maplab_config](https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/python/exporters/kalibr_maplab_config) exporter it does not properly convert the distortion type. Using radtan model kalibr by [default](https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/CameraCalibrator.py#L655-L663) will export a:
```yaml
distortion_model: radtan
```
while rovioli expects "Unknown distortion model: "radtan". Valid values are {none, equidistant, fisheye, **radial-tangential**}.". It was simple to change, but it would be nice to have the converter actually convert it to the right format, or a note added to the wiki.

Attached is the file that I converted from (renamed to txt instead of yaml so it could be uploaded to github).
[camchain-imucam-2017-11-15-11-05-41_0.txt](https://github.com/ethz-asl/maplab/files/1515242/camchain-imucam-2017-11-15-11-05-41_0.txt)

Also the [IMU file](https://github.com/ethz-asl/maplab/wiki/Sensor-Calibration-Format#imu_parameters_maplab) on the wiki is missing the "id" and "sensor_type" fields.

goldbattle, 2017-11-29:
It also doesn't seem to export the IMU file when you specify it.
```
rosrun kalibr kalibr_maplab_config --label testing --cam camchain-imucam-2017-11-15-11-05-41_0.yaml --out ncamera.yaml --imu imu-2017-11-15-11-05-41_0.yaml --imu-out imu.yaml 
```
IMU file is attached.
[imu-2017-11-15-11-05-41_0.txt](https://github.com/ethz-asl/maplab/files/1515382/imu-2017-11-15-11-05-41_0.txt)

fabianbl, 2017-11-30:
@goldbattle Thanks for the report, I will have a look.

fabianbl, 2017-11-30:
Will be fixed here: ethz-asl/kalibr#145.

Regarding the IMU file, you don't need to pass it to the converter. This is used for a different calibration format.

goldbattle, 2017-11-30:
I ended up being able to just not specify the `--to-ncamera` so it generated a single file with both the IMU and camera information. Needed to get the properly generated ID md5 hash and label (which I didn't realize that the label was the topic that the sensor is on).

fabianbl, 2017-12-01:
Actually you should provide separate calibration files for the IMU when using rovioli, see here: https://github.com/ethz-asl/maplab/wiki/Running-ROVIOLI-in-VIO-mode

goldbattle, 2017-12-01:
Yep, but to generate the IMU file I used the command [above](https://github.com/ethz-asl/maplab/issues/4#issuecomment-347983046) and copied the resulted combo file into the two separate files that are needed. Just tested the pull request and that fixed the radtan issue for me.

I still get errors on processing the IMU file.
Seems that the IMU parser is unable to handle the current kalibr IMU format output (multi-imu).
I fixed by simply ensuring that the first IMU is always selected
```python
def parse_from_kalibr_format(self, imu_dict):
        imu_dict = imu_dict['imu0'] #added this line since camchain is relative to imu0
        self.rostopic = imu_dict['rostopic']
```

Not the cleanest way to do things, but it worked for me.
[tempconfig.txt](https://github.com/ethz-asl/maplab/files/1521966/tempconfig.txt)

dymczykm, 2017-12-05:
@goldbattle I just successfully used the command from wiki to calibrate a new sensor, I used:
```
rosrun kalibr kalibr_maplab_config --to-ncamera --label zr300 --cam camchain-imucam-2017-12-05-10-52-08.yaml
```

goldbattle, 2017-12-05:
Yep, I was just noting that the kalibr IMU config (not the camera one) to maplab doesn't work as is, and requires copying the contents of the generated file (generated without the `--to-ncamera`) into the two separate files, along with the change I noted above.

gakkiHp, 2019-02-27:
@dymczykm 
can this：rosrun kalibr kalibr_maplab_config --to-ncamera --label zr300 --cam camchain-imucam-2017-12-05-10-52-08.yaml generate three yaml files that we need？
imu-zr300-maplab.yaml imu-zr300-rovio.yaml ncamera.yaml