ethz-asl/maplab #5: ROS Bag Image Topics

goldbattle, 2017-11-29:
Is there a way to remove the suffix from the camera topics?

For example when loading a bag the correct topic:
`/bb2/left/image_color`
Gets replaced with this incorrect one:
`/bb2/left/image_color/image_raw`
I just manually removed it [here](https://github.com/ethz-asl/maplab/blob/29b906cd55716beb2c33e32d43ca84a7abcc99ce/common/vio-common/src/rostopic-settings.cc#L11-L13) but I don't see a clear way to specify to use the correct topic in the config file.

andrepfr, 2017-11-30:
You should be able to just pass in an empty string for the vio_camera_topic_suffix flag. 
```
rosrun rovioli rovioli --vio_camera_topic_suffix="" [YOUR_OTHER_FLAGS]
```
If you are using one of the scripts you can do the following:
```
rosrun rovioli tutorial_euroc save_folder [SAVE_FOLDER] [BAG_FILE] --vio_camera_topic_suffix=""
```

goldbattle, 2017-11-30:
Sweet, I assume that this works for all the variables defined at the top of the files, would be nice to have a parameter table/list someplace. Thanks!

closed 2017-11-30 by goldbattle

mfehr, 2017-11-30:
@goldbattle thanks for your feedback. You can get a full list of all available parameters by adding `--help`  to any executable. However it's a very long list. We are currently working on a wiki page to help navigating all the gflags/parameters. For now I can tell you this: the flags are grouped by component/function using prefixes, e.g. vio_, lc_ for loop closure, ba_ for bundle adjustment,  etc.
In the maplab console you can then easily use the autocomplete function to get a list of relevant parameters.

xukuanHIT, 2018-01-03:
Hi , how to change the imu topic ? @mfehr

mfehr, 2018-01-03:
Hi @xukuanHIT, you can modify the imu topic by modifying [this line](https://github.com/ethz-asl/maplab/blob/master/applications/rovioli/share/imu-adis16488.yaml#L2) in your imu config file.
Example: 
```
hardware_id: imu0 
```
will map to
```
/imu0
```