ethz-asl/maplab #7: Build the online part of maplab for ARMv7-A (32bit)

Alabate, 2017-12-04:
First of all, thank you for this awesome project.

I'd like to use the online part of maplab on an embedded 32bit ARM cpu. I didn't find any way to build only the online part, so I tried to build the whole project and it fails on those errors:

* `maplab_common` assert 64bit systems:
```
Errors     << maplab_common:make /home/balyo/maplab/logs/maplab_common/build.make.001.log                                                                                                    
/home/balyo/maplab/src/maplab/common/maplab-common/src/unique-id.cc: In function 'void common::internal::generateUnique128BitHash(uint64_t*)':
/home/balyo/maplab/src/maplab/common/maplab-common/src/unique-id.cc:9:3: error: static assertion failed: Please adapt the below to your non-64-bit system.
   static_assert(
   ^
make[2]: *** [CMakeFiles/maplab_common.dir/src/unique-id.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/maplab_common.dir/all] Error 2
make: *** [all] Error 2
cd /home/balyo/maplab/build/maplab_common; catkin build --get-env maplab_common | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
```

* `rovio` use x86-specific option:
```
Errors     << rovio:make /home/balyo/maplab/logs/rovio/build.make.000.log                                                                                                                    
c++: error: unrecognized command line option '-mssse3'
make[2]: *** [CMakeFiles/rovio.dir/src/CameraCalibration.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```

How much work would be required to make it compatible with this architecture ?

dymczykm, 2017-12-05:
Hi, thanks a lot for your question.

To be honest, the amount of work is difficult to predict, but I think it's worth a try. It's also hard to say what will be the impact on performance as you're lacking SSE optimization that Eigen broadly uses (and Eigen matrices are present all over the project).

1) This Id problem in ``maplab_common`` is rather easy to fix, you just need to hack in a routine to generate a unique id on a 32-bit machine. As you can see, id is composed of two 64-bit values so you just need to properly initialize them (make sure the id space is 128bit not 64bit to avoid Id clashes).

2) ```-mssse3`` flag is just to enable Eigen optimizations using SSE3. You can probably kick out this flag for now and see what happens, I don't think we strictly depend on SSE, it was just to promote the optimization by the compiler.

Another issue you may encounter are Eigen alignment issues (we might be missing proper alignment in some places), which show up much more often on 32-bit machines. On the other hand, they are resulting from x86 SSE optimizations, not sure if Eigen has some ARM-specific code (NEON?) that might cause them.

Alabate, 2017-12-07:
Ok, so I managed to build and start a bag. I'm giving you the diff of my workspace, to let anyone do the same. But those are really dirty modifications, sometimes even useless or just dumb (like in `descriptor-projection` where I tried to add 3 times the same thing in the `CMakeLists.txt` because it wouldn't work). We have to find clean ways to fix the same problems.

For `src/maplab`:
```
diff --git a/algorithms/loopclosure/descriptor-projection/CMakeLists.txt b/algorithms/loopclosure/descriptor-projection/CMakeLists.txt
index 3653b75..9770ff0 100644
--- a/algorithms/loopclosure/descriptor-projection/CMakeLists.txt
+++ b/algorithms/loopclosure/descriptor-projection/CMakeLists.txt
@@ -4,11 +4,16 @@ project(descriptor_projection)
 find_package(catkin_simple REQUIRED)
 catkin_simple()
 
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mfpu=neon")
+add_definitions("-mfpu=neon")
+
 set(PROTO_DEFNS proto/descriptor-projection/projected_image.proto)
 PROTOBUF_CATKIN_GENERATE_CPP2("proto" PROTO_SRCS PROTO_HDRS ${PROTO_DEFNS})
 include_directories(${CMAKE_CURRENT_BINARY_DIR})
 
 set(LIBRARY_NAME ${PROJECT_NAME})
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mfpu=neon")
+add_definitions("-mfpu=neon")
 
 cs_add_library(${LIBRARY_NAME} src/build-projection-matrix.cc
                                src/descriptor-projection.cc
@@ -28,6 +33,8 @@ if(CATKIN_ENABLE_TESTING)
   target_link_libraries(test_matching_based_lc_quantizer_serialization ${LIBRARY_NAME})
 endif()
 
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mfpu=neon")
+add_definitions("-mfpu=neon")
 # CMake Indexing
 FILE(GLOB_RECURSE LibFiles "include/*")
 add_custom_target(headers SOURCES ${LibFiles})
diff --git a/algorithms/loopclosure/descriptor-projection/include/descriptor-projection/descriptor-projection.h b/algorithms/loopclosure/descriptor-projection/include/descriptor-projection/descriptor-projection.h
index 7fad532..9d974b0 100644
--- a/algorithms/loopclosure/descriptor-projection/include/descriptor-projection/descriptor-projection.h
+++ b/algorithms/loopclosure/descriptor-projection/include/descriptor-projection/descriptor-projection.h
@@ -67,20 +67,11 @@ void DescriptorToEigenMatrix(
 
 // Define a set of macros to NEON and SSE instructions so we can use the same
 // code further down for both platforms.
-#ifdef __ARM_NEON__
 #define VECTOR_SET vdupq_n_u8       // Set a vector from a single uint8.
 #define VECTOR_LOAD(x) vld1q_u8(x)  // Set a vector from a mem location.
 #define VECTOR_TYPE uint8x16_t      // The type of the vector element.
 #define VECTOR_AND vandq_u8         // The vector AND instruction.
 #define VECTOR_EXTRACT(x, i) vgetq_lane_u8(x, i)  // Get element from vector.
-#else
-#define VECTOR_SET _mm_set1_epi8
-#define VECTOR_LOAD(x) _mm_load_si128(reinterpret_cast<const __m128i*>(x))
-#define VECTOR_TYPE __m128i
-#define VECTOR_AND _mm_and_si128
-// Could use _mm_extract_epi8, but this requires SSE4.1.
-#define VECTOR_EXTRACT(x, i) reinterpret_cast<const char*>(&x)[i]
-#endif  // __ARM_NEON__
 
   VECTOR_TYPE mask[8];
   mask[0] = VECTOR_SET((1 << 0));
@@ -142,20 +133,11 @@ void DescriptorToEigenMatrix(
 
 // Define a set of macros to NEON and SSE instructions so we can use the same
 // code further down for both platforms.
-#ifdef ANDROID
 #define VECTOR_SET vdupq_n_u8       // Set a vector from a single uint8.
 #define VECTOR_LOAD(x) vld1q_u8(x)  // Set a vector from a mem location.
 #define VECTOR_TYPE uint8x16_t      // The type of the vector element.
 #define VECTOR_AND vandq_u8         // The vector AND instruction.
 #define VECTOR_EXTRACT(x, i) vgetq_lane_u8(x, i)  // Get element from vector.
-#else
[... 196 lines cut ...]
--- a/internal/ethzasl_brisk/brisk_ros_demo/CMakeLists.txt
+++ b/internal/ethzasl_brisk/brisk_ros_demo/CMakeLists.txt
@@ -4,7 +4,7 @@ project(brisk_ros_demo)
 find_package(catkin_simple REQUIRED)
 catkin_simple(ALL_DEPS_REQUIRED)
 
-add_definitions(-msse2 -mssse3 -march=native -Wextra -Wall
+add_definitions(-msse2 -march=native -Wextra -Wall
                 -Wno-overloaded-virtual -Wno-non-virtual-dtor # OpenCV.
                 -std=c++0x -DENABLE_BRISK_TIMING -DTEST_IN_SOURCE
                 -DHAVE_OPENCV)
Submodule internal/maplab_rovio contains modified content
diff --git a/internal/maplab_rovio/CMakeLists.txt b/internal/maplab_rovio/CMakeLists.txt
index b487b46..7ff35d8 100644
--- a/internal/maplab_rovio/CMakeLists.txt
+++ b/internal/maplab_rovio/CMakeLists.txt
@@ -1,6 +1,6 @@
 cmake_minimum_required (VERSION 2.6)
 project(rovio)
-set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -mssse3")
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
 set(ROVIO_NMAXFEATURE 25 CACHE STRING "Number of features for ROVIO")
 set(ROVIO_NCAM 1 CACHE STRING "Number of enabled cameras")
 set(ROVIO_NLEVELS 4 CACHE STRING "Number of image leavels for the features")
```

Alabate, 2017-12-07:
As you can see, most of modifications are removing the `-mssse3` to add `-mfpu=neon` instead. How could we do this in a clean way ?

* I see on some projects that there is an `ANDROID` flag in cmake that enable neon. We could use this flag in every part of the project. In this case a simple parameter in the catkin worskpace would enable the arm build. However, I don't think `ANDROID` is a good name for a such flag, because android is not the only arm system and android can be installed on x86.
* In `aslam_cv2/aslam_cv_common/cmake/export_flags.cmake` there is also the `IS_ARM_ARCHITECTURE` flag which is generated by the following script. This could also be a good solution, the name of the flag is definitely better. But in this case, we have to put this script somewhere and include it in every CMakeFile.txt. We could put it in every repo that need it, or in another simple repo which would be a dependency.
```
set(IS_ARM_ARCHITECTURE FALSE)
execute_process(COMMAND uname -m COMMAND tr -d '\n' OUTPUT_VARIABLE ARCHITECTURE)
if (ARCHITECTURE MATCHES "^(arm)")
  set(IS_ARM_ARCHITECTURE TRUE)
endif()
```

What do you think ?

fabianbl, 2017-12-07:
@Alabate Thanks a lot for your efforts to port maplab to ARM based devices and for sharing your diff. Regarding your remarks: I agree, `ANDROID` is not a good flag name for this purpose, but having a global cmake flag which is set based on the processor architecture sounds like a good idea. Potentially, we can also replace the script from `aslam_cv_common` with a combination of [FindARM.cmake](https://github.com/PX4/uvc_ros_driver/blob/master/cmake/FindARM.cmake) and [FindSSE.cmake](https://gist.github.com/hideo55/5642892) and then set the `-mssse3` or `-mpfu=neon` flags based on this. I think we can put this script just in `aslam_cv_common` and then copy it to the two other repos that also use it. As I'm currently also building maplab on an Odroid, I will try to put something together and tag you, ok?

fabianbl, 2017-12-07:
Actually a think [this](https://github.com/tunabrain/tungsten/blob/master/cmake/OptimizeForArchitecture.cmake) script might do the job.

bidbest, 2017-12-08:
Guys, thanks for the amazing project. As @Alabate I was trying to make it work on the Nvidia TX1. 
Like you point out, the extensive use of -mssse3 is the biggest concern.  Regarding the dependencies, I manage to compile:

doxygen_catkin   -> ok
catkin_simple      -> ok
gflags                  -> ok
glogs                   -> ok (use lates version from git,  remove the patch in the make, 
                                      have libunwind-dev  libunwind8-dev pre installed)
Voxblox               -> ok
Opencv3_catkin  -> ok

the rest still trying out, especially aslam pkg (-mssse3) and brisk. Brisk released a v2, and  it should implement NEON. Still need to try it.

Keep you update and thanks for the work!

Changliu52, 2017-12-08:
Could this be a simple fix - https://github.com/JakobEngel/dso/pull/59 ?

hoangthien94, 2017-12-15:
@bidbest Have you managed to build brisk on the TX1? When I remove the `-msse3` option I get error:
`
fatal error: emmintrin.h: No such file or directory
`
I found this [SSE2NEON.h](https://github.com/jratcliff63367/sse2neon/blob/master/SSE2NEON.h), maybe this could help?

bidbest, 2017-12-15:
@hoangthien94 Diden't try it yet. I'm currently working on x86 processors, will try again on arm in the future. If I remember correctly emmintrin.h file is considered only when flagh sse3 is present. You can try to test Brisk v2, that should support NEON., and hope it will not conflict with the rest of maplab.

fabianbl, 2017-12-15:
@hoangthien94 We are working on porting maplab to ARM. Regarding ethz-asl/brisk repo, you can alreay checkout the feature/port-to-arm branch there which should do the job.

megamindhe, 2017-12-17:
Hi @fabianbl, I have tried to checkout the feature/port-to-arm branch ,but still failed when building brisk with following errors:
```shell
c++: error: unrecognized command line option ‘-mfpu=neon’
```
My device is Nvidia Jetson TX2 which's architecture is aarch64.
```shell
$ uname -a
Linux CAR01-CU03 4.4.38 #1 SMP PREEMPT Tue Oct 10 15:24:32 CST 2017 aarch64 aarch64 aarch64 GNU/Linux
```

fabianbl, 2017-12-17:
Hi @megamindhe . Thank you a lot for your report. Unfortunately, we weren't able yet to test this on an `armv8` architecture (but will do so soon). Apparently, for these processors, no specifications for using advanced SIMD are necessary (see [here](https://stackoverflow.com/a/29891469)). For now, you can simply comment out all occurrences of `-mfpu=neon` (e.g. [here](https://github.com/ethz-asl/ethzasl_brisk/blob/feature/port-to-arm/agast/cmake/export_flags.cmake#L6) in ethz-asl/ethzasl_brisk).

fabianbl, 2018-01-24:
We released an experimental pre-release branch which adds build compatibility for `armv7` and `armv8` processors (along with some other changes).

You can test it by checking out the branch `pre-release-january` in both the `maplab` and the `maplab_dependencies` repo:

```bash
git pull
git checkout pre-release-january
git submodule update --recursive
```

After this, clean the catkin workspace and rebuild it:

```bash
catkin clean
catkin build maplab
```

We are happy to hear if this works for you.

dan9thsense, 2018-01-30:
I tried building on a Jetson tx2, ubuntu 16.04, ROS kinetic, with the experimental pre-release branch.
The initial try failed with an "internal compiler error" while building rovio.  Looking at the logfile, the error occurred right after:

> [ 94%] Built target feature_tracker_node

On the second try, it succeeded with rovio but failed while building vi_map (46/91), with this error:

> Errors     << vi_map:cmake /home/nvidia/maplab_ws/logs/vi_map/build.cmake.000.log                                                                                                              
> CMake Error at /home/nvidia/maplab_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:38 (find_package):
>   By not providing "Findbenchmark_catkin.cmake" in CMAKE_MODULE_PATH this
>   project has asked CMake to find a package configuration file provided by
>   "benchmark_catkin", but CMake did not find one.
> 
>   Could not find a package configuration file provided by "benchmark_catkin"
>   with any of the following names:
> 
>     benchmark_catkinConfig.cmake
>     benchmark_catkin-config.cmake

Here is the output of catkin config:

> nvidia@tegra-ubuntu:~/maplab_ws$ catkin config
> -----------------------------------------------------------
> Profile:                     default
> Extending:        [explicit] /opt/ros/kinetic
> Workspace:                   /home/nvidia/maplab_ws
> -----------------------------------------------------------
> Source Space:       [exists] /home/nvidia/maplab_ws/src
> Log Space:          [exists] /home/nvidia/maplab_ws/logs
> Build Space:        [exists] /home/nvidia/maplab_ws/build
> Devel Space:        [exists] /home/nvidia/maplab_ws/devel
> Install Space:      [unused] /home/nvidia/maplab_ws/install
> DESTDIR:            [unused] None
> -----------------------------------------------------------
> Devel Space Layout:          merged
> Install Space Layout:        None
> -----------------------------------------------------------
> Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
> Additional Make Args:        None
> Additional catkin Make Args: None
> Internal Make Job Server:    True
> Cache Job Environments:      False
> -----------------------------------------------------------
> Whitelisted Packages:        None
> Blacklisted Packages:        None
> -----------------------------------------------------------
> Workspace configuration appears valid.
> -----------------------------------------------------------

fabianbl, 2018-01-30:
@dan9thsense Thank you for your report. I think what you can do is:

```bash
cd ~/maplab_ws/src/maplab_dependencies
git submodule init
git submodule update
```

This will initialize and checkout benchmark_catkin which was not there in the master branch. Forgot to mention this, I now added it to the description above.

mfehr, 2018-01-30:
@fabianbl I added your instructions to the PR description as well, thx.

dan9thsense, 2018-01-30:
Good news!  With that change the entire package builds successfully.

closed 2018-08-15 by mfehr