ethz-asl/maplab #8: Build Error: build opencv3_catkin failed

xuqingwenkk, 2017-12-04:
Hi, 
I met a problem when execute `catkin build maplap`. The following error occurs:
`make[2]: *** [opencv3_contrib-prefix/src/opencv3_contrib-stamp/opencv3_contrib-download] Error 1
make[1]: *** [CMakeFiles/opencv3_contrib.dir/all] Error 2
make: *** [all] Error 2
cd /home/xuqw/maplab_ws/build/opencv3_catkin; catkin build --get-env opencv3_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -` 
I found that there is an answer on FAQ: 
[https://github.com/ethz-asl/maplab/wiki/FAQ#q-why-is-my-opencv3_catkin-build-freezing](url), 
but I didn't installed matlab.
Could you tell me other possible reasons?
Best wishes
Qingwen

goldbattle, 2017-12-04:
Did you have access to the internet when you built this?
It will download opencv from a github repo, I would make sure that you have internet access.
Are there any other errors, there isn't that much to go off of with that error message?

dymczykm, 2017-12-04:
As @goldbattle suggested, please retry making sure your internet connection is stable. Additionally, you might consider clearing your catkin workspace using ``catkin clean``.

The answer in FAQ shouldn't be related to the problem you're observing.

xuqingwenkk, 2017-12-24:
Solved! Network Problem. Thanks!

closed 2017-12-24 by xuqingwenkk