ethz-asl/maplab #9: catkin build maplab failed

wendaoyuchen, 2017-12-05:
Hi, I want to try maplab on my computer(Ubuntu16.04 + ROS kinetic),
but when I run _catkin build maplab_in the directory "catkin_ws", it is shown as below:
_The build space at '/home/catkin_ws/build' was previously built by 'catkin_make'. Please remove the build space or pick a different build space._
How could I solve this problem?

dymczykm, 2017-12-05:
It seems that you previously used ``catkin make`` in the same workspace. Could you please call ``catkin clean`` and try again?

wendaoyuchen, 2017-12-06:
@dymczykm Thanks a lot, but maybe I am not familiar with ROS ,when I run `catkin clean`, it reports another error message again,
_[clean] Error: The current or desired workspace could not be determined. Please run `catkin clean` from within a catkin workspace or specify the workspace explicitly with the `--workspace` option._
So I will try maplab on the other computer, thanks a lot again.

goldbattle, 2017-12-06:
You should init a workspace (`catkin init`), or ensure that you are in the folder that you original created.
That error is noting that you are not in a folder that is a catkin workspace, so I would make sure you cd into the folder that should have a `src` `devel` etc folders in it.

wendaoyuchen, 2017-12-06:
@goldbattle OK, thank you. I built a new catkin_ws workspace, and it works. But another problem comes to me.
`Failed     << ceres_catkin:make `
I guess I miss the ceres_src, but I do not know how to build it?

eggerk, 2017-12-06:
Do you have the full error message? One possible reason is that you had no or unstable internet during compilation. `ceres_catkin` (and other dependencies) will download the source code during compilation.

wendaoyuchen, 2017-12-06:
@eggerk Thanks a lot, I found the error as you said, so the unstable Internet stops my road to Roman.

qinyusen, 2017-12-29:
@wendaoyuchen   what you need is just a proxy tools like proxychains to reach the google's repo, if you now in Mainland of P.R.C.

mfehr, 2018-01-12:
Closing this issue. We added a [FAQ entry](https://github.com/ethz-asl/maplab/wiki/FAQ#q-i-have-problems-building-the-opencv3-ceres-eigen-protobuf_catkin-package) and a section in [Known Issues](https://github.com/ethz-asl/maplab/wiki/Known-Issues#building-with-unstable-internet-connection).

closed 2018-01-12 by mfehr

QinZiwen, 2018-07-24:
After install ROS-kinetic, when push tab key:
```
$ catkin_
catkin_create_pkg          catkin_make                catkin_test_changelog
catkin_find                catkin_make_isolated       catkin_test_results
catkin_find_pkg            catkin_package_version     catkin_topological_order
catkin_generate_changelog  catkin_prepare_release     
catkin_init_workspace      catkin_tag_changelog
```
There is no catkin, when I init a workspace:
```
~/catkin_ws$ catkin init
catkin: command not found
```

ZacharyTaylor, 2018-07-24:
The `catkin` command is part of the python-catkin-tools package and is installed with `sudo apt install python-catkin-tools`. If you are running into this issue you are likely also missing other deps as well and you may need to go through the [install instructions](https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu) again.

qinyusen, 2018-07-24:
@QinZiwen  RTFM.

Obviously, you need to try to learn some basic Linux knowledge.  You lost some instruction or install failed when you using “apt” install.

Chinese Version below:

好好学学linux吧，你问的问题是说，根本就没找到catkin这个工具。 这个基本上是linux初学者，特别初级的那种问题，所以，我推荐你先学一下linux基础知识。

回到问题，说白了就是没装上工具， 在安装指令的那个apt那一行，如果那一行失败，往往是因为：1.你ubuntu版本用错了，2 有可能你换了国内的源，部分国内的源有问题装不了ros。

你可以依次debug，之后，因为你catkin没装上，我理解对应下面这行里的某些工具你也应该没装上。我推荐你单独执行一下，好好看看那些安装失败了，然后查一下为什么失败了，然后debug一下。

“sudo apt install autotools-dev ccache doxygen dh-autoreconf git liblapack-dev libblas-dev libgtest-dev libreadline-dev libssh2-1-dev pylint clang-format-3.8 python-autopep8 python-catkin-tools python-pip python-git python-setuptools python-termcolor python-wstool libatlas3-base --yes”

最后，你这种问题很不礼貌，甚至有些无理取闹。有点儿像你的电脑没装操作系统，然后你在这儿问为什么你的电脑不能用来上网。

QinZiwen, 2018-07-25:
@ZacharyTaylor @qinyusen Thanks.

QinZiwen, 2018-07-25:
How much memory is needed to compile maplab？ I only have 4GB of memory and intel i3 4-core CPU, Ubuntu stuck when compiled to opencv3_catkin。

QinZiwen, 2018-07-25:
I adjusted the number of jobs at compile time and the problem was solved.

phil123456, 2020-03-10:
on melodic,
even after catkin init, I still have the same error
catkin clean or catkin_make clean did not solve the issue

936804292, 2020-09-19:
> @QinZiwen RTFM.
>
> Obviously, you need to try to learn some basic Linux knowledge. You lost some instruction or install failed when you using “apt” install.
>
> Chinese Version below: [...]

你牛逼我们就不会在这相遇了:)

Guo-lab, 2024-01-01:
> @dymczykm Thanks a lot, but maybe I am not familiar with ROS ,when I run `catkin clean`, it reports another error message again, _[clean] Error: The current or desired workspace could not be [...]

This might result from a lack of `source devel/setup.bash`.