ethz-asl/maplab #12: Failure to compile in Nvidia TX2 (ARM)

Changliu52, 2017-12-06:
Hi,
I tried to compile the maplab following the standard steps in wiki.
My environment is Nvidia TX2, running ubuntu 16.04 (flashed using Jetpack 3.1), ROS Kinetic.
It failed with the error:
```
cd /home/nvidia/ws_maplab/build/opengv; catkin build --get-env opengv | catkin env -si  /usr/bin/cmake /home/nvidia/ws_maplab/src/maplab_dependencies/3rdparty/opengv --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/nvidia/ws_maplab/devel -DCMAKE_INSTALL_PREFIX=/home/nvidia/ws_maplab/install -DCMAKE_BUILD_TYPE=Release; cd -
...............................................................................
_______________________________________________________________________________
Errors     << glog_catkin:make /home/nvidia/ws_maplab/logs/glog_catkin/build.make.000.log
/home/nvidia/ws_maplab/build/glog_catkin/glog_src-prefix/src/glog_src/missing: Unknown `--is-lightweight' option
Try `/home/nvidia/ws_maplab/build/glog_catkin/glog_src-prefix/src/glog_src/missing --help' for more information
configure: WARNING: 'missing' script is too old or missing
./config.guess: unable to guess system type

This script, last modified 2007-07-22, has failed to recognize
the operating system you are using. It is advised that you
download the most up to date version of the config scripts from

  http://savannah.gnu.org/cgi-bin/viewcvs/*checkout*/config/config/config.guess
and
  http://savannah.gnu.org/cgi-bin/viewcvs/*checkout*/config/config/config.sub

If the version you run (./config.guess) is already up to date, please
send the following data and any information you think might be
pertinent to <config-patches@gnu.org> in order to provide the needed
information to handle your system.

config.guess timestamp = 2007-07-22

uname -m = aarch64
uname -r = 4.4.38+
uname -s = Linux
uname -v = #7 SMP PREEMPT Tue Nov 7 12:35:49 EST 2017

/usr/bin/uname -p = 
/bin/uname -X     = 

hostinfo               = 
/bin/universe          = 
/usr/bin/arch -k       = 
/bin/arch              = 
/usr/bin/oslevel       = 
/usr/convex/getsysinfo = 

UNAME_MACHINE = aarch64
UNAME_RELEASE = 4.4.38+
UNAME_SYSTEM  = Linux
UNAME_VERSION = #7 SMP PREEMPT Tue Nov 7 12:35:49 EST 2017
configure: error: cannot guess build type; you must specify one
make[2]: *** [glog_src-prefix/src/glog_src-stamp/glog_src-configure] Error 1
make[1]: *** [CMakeFiles/glog_src.dir/all] Error 2
make: *** [all] Error 2
```

The compilation on i5 ubuntu16.04-kinetic works perfectly.
Seems the arm architecture is completely not supported?
Or is there anyway to only compile ROVIOLI on the arm?

Thank you in advance.
Kind Regards,
Chang

Alabate, 2017-12-07:
Hi, I manage to built on ARM, see #7 . But I didn't get this `unable to guess system type`. If you manage to go through this error, the diff I gave may help you, even if we don't exactly have the same target architecture.

bidbest, 2017-12-08:
I installed glogs on the TX1.
install (apt)  libunwind-dev and libunwind8-dev. Then use this CMakeList.txt  

``` cmake
cmake_minimum_required(VERSION 2.8.3)
project(glog_catkin)

find_package(catkin_simple REQUIRED)
catkin_simple()

include(ExternalProject)

file(MAKE_DIRECTORY ${CATKIN_DEVEL_PREFIX}/include)

set(VERSION 0.3.5)
ExternalProject_Add(glog_src
  #URL https://github.com/google/glog/archive/v${VERSION}.zip
  GIT_REPOSITORY https://github.com/google/glog.git
  UPDATE_COMMAND ""
#  PATCH_COMMAND patch -p0 < ${CMAKE_SOURCE_DIR}/fix-unused-typedef-warning.patch
  CONFIGURE_COMMAND cd ../glog_src/ && ./autogen.sh && \\
    ./configure --with-pic
    --with-gflags=${gflags_catkin_PREFIX}
    --prefix=${CATKIN_DEVEL_PREFIX}
  BUILD_COMMAND cd ../glog_src/ && make -j 8
  INSTALL_COMMAND cd ../glog_src/ && make install -j 8
)

cs_add_library(${PROJECT_NAME} src/dependency_tracker.cc)
add_dependencies(${PROJECT_NAME} glog_src)
target_link_libraries(${PROJECT_NAME} ${CATKIN_DEVEL_PREFIX}/lib/libglog${CMAKE_SHARED_LIBRARY_SUFFIX})

cs_install()

cs_export(INCLUDE_DIRS ${CATKIN_DEVEL_PREFIX}/include
  CFG_EXTRAS glog-extras.cmake)
```

This simply download the lastest version of glog. This new version needs a previous command (autogen.ssh) before the .configure.
 Also removes the patch.

Changliu52, 2017-12-08:
Great! Thank you @bidbest . It worked with the new CMakeList. 
Now It failed with the error: unrecognised command line option '-mssse3'. So I guess I will just need to implement @Alabate 's diff. 
Thanks again.

closed 2017-12-08 by Changliu52

fabianbl, 2017-12-18:
Hi @Changliu52. I guess you should be fine by removing all `-mssse3` / `-mfpu=neon` compiler instructions in the code, see #18. I hope we can provide an update soon.

fabianbl, 2018-01-24:
@Changliu52 The `pre-release-january` branch in both the `maplab` and the `maplab_dependencies` repo adds build compatibility for ARM processors, see here: ethz-asl/maplab#7

meitiever, 2020-03-03:
I deleted all mssse3 definitions in all CMAKELIST.txt. and all the deps are built correctly.
but the brisk package does not support neon and arm does not support mssse3.
so you can't build maplab on arm.
for a developer, it is quite easy to fix this - by replacing brisk with OpenCV.
and then I think maplab can be run on an arm device.
On the other hand, the performance issue is quite big.
because the maplab does not build on CUDA.
it require quite a lot of work to make this work on CUDA smoothly.