ethz-asl/maplab #13: Failed on CSV export

MLPhD, 2017-12-11:
Hi,
when I try to export the CSV file from the map, an error appears and no data are exported:
"F1211 18:56:26.139300  6256 channel-definitions.h:21] Check failed: it != channels.end() Channelgroup does not contain channel VISUAL_KEYPOINT_SCALES".
Thank you for the help,
Michela

eggerk, 2017-12-12:
Hi Michela,

thanks for your report.

Could you send us the parameters you used to create the map? I couldn't reproduce this on my machine, but this does look like a bug. We will provide a fix for this soon.

MLPhD, 2017-12-12:
Hi Kevin,

Yes, sure!

I created the map as explain in the sample of
https://github.com/ethz-asl/maplab/wiki/Console-map-management , loading
the map included in the test data repository
<https://github.com/ethz-asl/maplab_test_data/tree/master/test_maps/vi_app_test>
.
Then I visualized the map in rviz.
In maplab console I tried to create the csv data files with this command:

   - maplab <vi_app_test>:/$ csv_export --csv_export_path
   ~/Scrivania/csv_data

And what I obtained is:

   - I1211 18:56:26.138227  6256 csv-export.cc:238] Exporting data from
   mission c795..b551.
   F1211 18:56:26.139300  6256 channel-definitions.h:21] Check failed: it
   != channels.end() Channelgroup does not contain channel
   VISUAL_KEYPOINT_SCALES
   *** Check failure stack trace: ***
   *** Aborted at 1513014986 (unix time) try "date -d @1513014986" if you
   are using GNU date ***
   PC: @     0x7fd98012d428 gsignal
   *** SIGABRT (@0x3e800001870) received by PID 6256 (TID 0x7fd98126dac0)
   from PID 6256; stack trace: ***
       @     0x7fd98012d4b0 (unknown)
       @     0x7fd98012d428 gsignal
       @     0x7fd98012f02a abort
       @           0x402973 common::glogFailFunctionWithConsoleRecovery()
       @     0x7fd980c7126d google::LogMessage::Fail()
       @     0x7fd980c73083 google::LogMessage::SendToLog()
       @     0x7fd980c70dfb google::LogMessage::Flush()
       @     0x7fd980c73a6e google::LogMessageFatal::~LogMessageFatal()
       @     0x7fd97c72408e
   aslam::channels::get_VISUAL_KEYPOINT_SCALES_Data()
       @     0x7fd96b1910dd
   _ZNSt17_Function_handlerIFvjRKN5aslam11VisualFrameEEZN10csv_export12_GLOBAL__N_128exportVerticesAndTracksToCsvERKN6vi_map5VIMapERKNS7_9MissionIdERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEPmPSt13unordered_mapIN10pose_graph8VertexIdEmSt4hashISP_ESt8equal_toISP_ESaISt4pairIKSP_mEEEEUljS3_E_E9_M_invokeERKSt9_Any_dataOjS3_
       @     0x7fd97ce8e33f vi_map::Vertex::forEachFrame()
       @     0x7fd96b192b2a csv_export::(anonymous
   namespace)::exportVerticesAndTracksToCsv()
       @     0x7fd96b194051 csv_export::exportMapToCsv()
       @     0x7fd96b3b5f92
   _ZNSt17_Function_handlerIFivEZN18data_import_export22DataImportExportPluginC4EPN6common7ConsoleEEUlvE_E9_M_invokeERKSt9_Any_data
       @     0x7fd980ea9ce4 common::CommandRegisterer::processCommand()
       @     0x7fd980eadc64 common::Console::RunCommand()
       @     0x7fd980eb06b4 common::Console::RunCommandPrompt()
       @           0x4026b1 main
       @     0x7fd980118830 __libc_start_main
       @           0x402739 _start
   Annullato (core dump creato)

The command creates a folder called ""csv_data" in my "Scrivania" (Desktop)
, but inside the folder there is only the "descriptor.csv" which is empty,
the "tracks.cvs" and "Vertices.csv" whit few data (I attached them in the
email).

If you need some other information, let me know.
Thank you for all!

Best,
Michela

-- 

-----------------------------------------------------
Ing. Michela Longhi

Ph.D Student
University of Rome "Tor Vergata"
Via del Politecnico 1, 00133 Rome, Italy
mob. +39 348 7455683
tel. +39 06 72597335
email: michela.longhi@uniroma2.it

Pervasive Electromagnetics Lab
----------------------------------------------------

eggerk, 2017-12-12:
Thanks for your response.

It's probably best not to use this test map as it is quite old and not all data might be in that map.

For now, to test maplab, I would recommend using one of the VI maps that we put in the wiki: https://github.com/ethz-asl/maplab/wiki/Sample-Datasets

MLPhD, 2017-12-12:
> maplab/wiki/Sample-Datasets

Thank for the suggestion,

I will try with this map.

Best,

Michela

-- 

-----------------------------------------------------
Ing. Michela Longhi

Ph.D Student
University of Rome "Tor Vergata"
Via del Politecnico 1, 00133 Rome, Italy
mob. +39 348 7455683
tel. +39 06 72597335
email: michela.longhi@uniroma2.it

Pervasive Electromagnetics Lab
----------------------------------------------------

MLPhD, 2017-12-12:
No it seems to run, also the csv export command!

Thanks

Michela

2017-12-12 10:02 GMT+01:00 Michela Longhi <michela.longhi1@gmail.com>:

> Thank for the suggestion,
>
> I will try with this map.
>
> Best,
>
> Michela
>
>
>> Thanks for your response.
>>
>> It's probably best not to use this test map as it is quite old and not
>> all data might be in that map.
>>
>> For now, to test maplab, I would recommend using one of the VI maps that
>> we put in the wiki: https://github.com/ethz-asl/ma
>> plab/wiki/Sample-Datasets
>>
>>
>
>
>
>
> -----------------------------------------------------
> Ing. Michela Longhi
>
> Ph.D Student
> University of Rome "Tor Vergata"
> Via del Politecnico 1, 00133 Rome, Italy
> mob. +39 348 7455683 <+39%20348%20745%205683>
> tel. +39 06 72597335 <+39%2006%207259%207335>
> email: michela.longhi@uniroma2.it
>
> Pervasive Electromagnetics Lab
> ----------------------------------------------------

-- 

-----------------------------------------------------
Ing. Michela Longhi

Ph.D Student
University of Rome "Tor Vergata"
Via del Politecnico 1, 00133 Rome, Italy
mob. +39 348 7455683
tel. +39 06 72597335
email: michela.longhi@uniroma2.it

Pervasive Electromagnetics Lab
----------------------------------------------------

ShrutheeshIR, 2022-05-15:
Hello, I am facing this issue as well, with my collected data (from a realsense). Any tips on how to solve it?
Thanks