ethz-asl/maplab #14: How to read and develop the code?

zhanghanduo, 2017-12-13:
Hi,
This whole package is really complicated and it will take me a long time for me to read through the code. But I don't know how to import such catkin_tools to an IDE. So I wonder using what tool you are developing the catkin_tools code? I had been using catkin make based package previously, so please forgive my ignorance of this experience. Thank you!

schneith, 2017-12-13:
That is a good point you are bringing up. With a project of this size an IDE with a working indexer is almost a must to navigate efficiently through the code. Most of us are using Eclipse and there are two different ways you can import the project:

1) **Importing the entire maplab repo as a single project** (preferred for maplab):
  https://github.com/ethz-asl/maplab/wiki/Importing-maplab-to-Eclipse

2) **Importing individual catkin packages as projects**:
  https://github.com/ethz-asl/eclipsify

The benefit of 1 is that only a single project has to be imported but no build/test targets will be created automatically. The method 2 creates build targets so that you can build from eclipse but each packages needs to be imported manually.

zhanghanduo, 2017-12-20:
Thank you for the answer! I chose the option 1 and now I am able to navigate the code. But I have to admit I am not quite used to Eclipse, after a long time use of Clion. Eclipse is quite lagging and sometimes even no response.

closed 2017-12-20 by zhanghanduo