ethz-asl/maplab #15: error: static assertion failed: Voxblox 3D points and maplab 3D points are not the same!

wendaoyuchen, 2017-12-13:
Hi, when I run` catkin build maplab`,this error comes out, How should I do to solve it?
 maplab/interfaces/voxblox_interface/src/integration.cc:291:3: error: static assertion failed: Voxblox 3D points and maplab 3D points are not the same!
 static_assert(
   ^
/maplab/interfaces/voxblox_interface/src/integration.cc:294:3: error: static assertion failed: Voxblox 3D point vectors and maplab 3D point vectors are not the same!
   static_assert(
   ^
/maplab/interfaces/voxblox_interface/src/integration.cc:297:3: error: static assertion failed: Voxblox transformations and maplab transformations are not the same!
   static_assert(

wendaoyuchen, 2017-12-14:
I find someone' solution to this problem, change the type` float` to `double` in voxblox package,
`typedef float FloatingPoint;` -> `typedef double FloatingPoint;`
in 41 line in 
`$CATKIN_WS/src/maplab_dependencies/internal/voxblox/voxblox/include/voxblox/core/common.h`

dymczykm, 2017-12-14:
@wendaoyuchen thanks for your report!

It seems that you have checked out a commit of voxblox that is incompatible with maplab. We have a package called ``maplab_dependencies`` that contains all the correct dependencies as submodules:
https://github.com/ethz-asl/maplab_dependencies/tree/master
and voxblox is specifically here:
https://github.com/ethz-asl/maplab_dependencies/tree/master/internal

To avoid any further problems, I would suggest to use ``maplab_dependencies`` to clone the correct versions of the dependencies. You can just follow the command from the [Installation wiki page](https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu):
```
git clone git@github.com:ethz-asl/maplab_dependencies.git --recursive
```

You can also go to ``maplab_dependencies/internal/voxblox`` to checkout the right commit:
```
git checkout 19634c9cb98efececbd8ae63d17e70d4f4594235
```