ethz-asl/maplab #16: Only an empty resource folder in save_folder after running ROVIOLI ?

pangfumin, 2017-12-14:
Hi,
 Thandk you for sharing this great slam framework !
When I run rovioli according to the toturial using rosrun, only an empty folder named resource can be found after the rosbag MH_01_easy.bag finish, and the rovioli can not exit from terminal. Is there someone meet the same situation? 
Can you help me for some advice?
Thank you!

dymczykm, 2017-12-14:
Hi @pangfumin,
thanks a lot for your report. We would like to replicate your problem to fix it, but we would need a few more details:
1) What is the exact command and the script if you use it? Could you please post both of them here?
2) Is the process terminating on its own, crashing, getting stuck, you kill it or call ctrl+c? Have you tried to wait a bit longer?
3) How long is the rosbag you are processing? Is it one of our sample datasets?
4) What is the path you specify for the output map (LOCALIZATION_MAP_OUTPUT in the script in tutorials)?
5) To rule out the possibility of a deadlock of the threads could you re-run the dataset with the following flag: ``-num_hardware_threads=1``

Just as a side-note. The warning that you're getting at the end of your output is not indicating a problem and we will soon get rid of it.
```
W1214 12:06:20.426757 1301 imu-camera-synchronizer.cc:103] No IMU messages have been received in the last 5 seconds. Check for measurement drops or if the topic is properly set in the maplab IMU configuration file
```

pangfumin, 2017-12-15:
Hi @dymczykm ,
Thank you for your timely reply!
1. I run rovioli using command  using the MH_01_easy.bag
> rosrun rovioli tutorial_euroc save_folder MH_01_easy.bag
2. The process get stuck, the image window of feature tracking of rovioli is also stuck, cannot use ctrl-c to kill it in terminal. when the rosbag get finished, after priting 
>  Retriangulating landmarks of 1827 vertices.

the warning start to show out
> W1214 12:06:05.423918 1301 imu-camera-synchronizer.cc:103] No IMU messages have been received in the last 5 seconds. Check for measurement drops or if the topic is properly set in the maplab IMU configuration file

I have try to wait for  a while , but the the vi-map saving process seems not going on, and the warning is endless.

3. I just use tutorial_euroc script to run the node , LOCALIZATION_MAP_OUTPUT is ./save_folder.

4. I have just try -num_hardware_threads=1 option , seems not help.

By the way , I am using ubuntu 14.04  ros: indigo. Eigen 3.3.4.
Thank you for your help.

mfehr, 2017-12-15:
Hi @pangfumin 

Thank you for providing us with the additional data.
We believe we managed to reproduce this issue on one (but only one) of our 14.04 machines.
What we observed is the following:
 * ROVIO (the window with the features) only seems to process about ~1 frame per second
 * The queues are filling up (can only be seen with --v=1)
 * At some point there are timeout warnings
 * After reaching the end of the rosbag we observed the same behavior you did, i.e. only the imu-cam-synchronizer seems to survive and it can't be killed with Ctr+C

If you could answer a couple of questions, that'd be very helpful:
 * Is this the same behavior you experienced, was ROVIO(LI) equally slow for you?
 * What gcc version are you using? 
 * Do you use a new catkin workspace or do you use an existing workspace? (i.e. are there other packages in your workspace?)
 * Could you rerun rovioli with the --v=1 flag and send us the glog output folder?
   ```bash
   mkdir logs
   rosrun rovioli tutorial_euroc save_folder MH_01_easy.bag --v=1 --log_dir=logs
   ```

pangfumin, 2017-12-16:
Hi @mfehr ,
Thank you for  providing these tips  :)

1. My gcc version is 4.8.4.
2. On my laptop,  the ROVIO is running at a normal frame rate, it is near real-time. This may be different from your recurrence.
3.  Maplab and maplab_dependencies are the only two ROS stack built in a new ROS workspace, using catkin build system. And they a built with -DCMAKE_BUILD_TYPE=Release.
4.  I use this scrip to try recording logs
>LOCALIZATION_MAP_OUTPUT=$1
ROSBAG=$2
NCAMERA_CALIBRATION="$ROVIO_CONFIG_DIR/ncamera-euroc.yaml"
IMU_PARAMETERS_MAPLAB="$ROVIO_CONFIG_DIR/imu-adis16488.yaml"
IMU_PARAMETERS_ROVIO="$ROVIO_CONFIG_DIR/imu-sigmas-rovio.yaml"
REST=$@
rosrun rovioli rovioli \
  --alsologtostderr=1 \
  --v=1 \
  --ncamera_calibration=$NCAMERA_CALIBRATION  \
  --imu_parameters_maplab=$IMU_PARAMETERS_MAPLAB \
  --imu_parameters_rovio=$IMU_PARAMETERS_ROVIO \
  --datasource_type="rosbag" \
  --save_map_folder="$LOCALIZATION_MAP_OUTPUT" \
  --optimize_map_to_localization_map=false \
  --map_builder_save_image_as_resources=false \
  --datasource_rosbag=$ROSBAG $REST \
  --log_dir = /home/pang/maplab_ws/maplablogs

but,   after  finishing the rosbag , it showing 
>**I1216 14:39:31.282711 32422 datasource-rosbag.cc:194] Rosbag playback finished!
I1216 14:39:31.531071 32381 rovioli-node.cc:110] Closing data source...
I1216 14:39:31.531124 32381 map-builder-flow.cc:111] Initializing landmarks of created map.
I1216 14:39:31.745971 32381 landmark-triangulation.cc:248] Getting vertices of mission: b939..0000
I1216 14:39:31.749847 32381 landmark-triangulation.cc:253] Retriangulating landmarks of 1736 vertices.
Progress: 0:  0.0% [---------] 1:  0.0% [---------] 2:  0.0% [---------] 3:  0.2% [---------]       Could not create logging file: No such file or directory
COULD NOT CREATE A LOGGINGFILE 20171216-143938.32381!W1216 14:39:38.327376 32409 imu-camera-synchronizer.cc:103] No IMU messages have been received in the last 5 seconds. Check for measurement drops or if the topic is properly set in the maplab IMU configuration file**
W1216 14:39:38.327689 32409 imu-camera-synchronizer.cc:111] No camera messages have been received in the last 5 seconds. Check for measurement drops or if the topic is properly set in the camera configuration file

mfehr, 2017-12-18:
Hi @pangfumin 
 * Looks like we were unable to reproduce your problem after all, also, the issue I described above vanished after rebooting, cleaning the workspace and rebuilding maplab. Could you give it a try, if you haven't done this already?
   ```bash
   # reboot
   # clean and rebuild
   cd ${CATKIN_WS}
   catkin clean --yes
   catkin build maplab
   ```
 * Did you create the log folder first? Logging fails if the folder doesn't exist. Could you give it another try?

pangfumin, 2017-12-25:
Hi， @mfehr, @dymczykm 
Thank you for your advise, clean the whole workspace solve this problem!
You can feel free to close this issue, thank you again.

mfehr, 2017-12-27:
@pangfumin thank you for your patience and cooperation. Glad it worked.

closed 2017-12-27 by mfehr