ethz-asl/maplab #18: Failure to compile in Nvidia TX2 (Aarch64)

megamindhe, 2017-12-17:
**First of all, thank you for this awesome project.**

I'm tring to compile the maplab On Nvidia Jetson TX2 which's architecture is aarch64.

- First, I tried [Alabate/maplab.diff](https://gist.github.com/Alabate/c1017c4f0a6c3b9e5e6db75ddbcddf0b#file-maplab_dependencies-diff-L78),it failed with following error when building brisk:
```shell
c++: error: unrecognized command line option ‘-mfpu=neon’
```

<br/><br/>
- The next ,I checked out the feature/port-to-arm of the brisk,but it still failed when building brisk with following errors:
```shell
 c++: error: unrecognized command line option ‘-mssse3’
make[2]: *** [CMakeFiles/rovio.dir/src/CameraCalibration.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
c++: error: unrecognized command line option ‘-mssse3’
make[2]: *** [CMakeFiles/rovio.dir/src/Camera.cpp.o] Error 1
c++: error: unrecognized command line option ‘-mssse3’
make[2]: *** [CMakeFiles/rovio.dir/src/FeatureCoordinates.cpp.o] Error 1
```

<br/><br/>
- Once again,I modified maplab_dependencies/internal/ethzasl_brisk/agast/cmake/export_flags.cmake to:
```shell
set(HAS_NEON TRUE)
add_definitions(-mfpu=neon)
```
It failed when building brisk with following errors:
```shell
c++: error: unrecognized command line option ‘-mfpu=neon’
```

<br/><br/>
- Finally,I modified maplab_dependencies/internal/ethzasl_brisk/agast/cmake/export_flags.cmake to:
```shell
set(HAS_NEON OFF)
#add_definitions(-mfpu=neon) 
```
It failed when building brisk with following errors:
```shell
maplab_dependencies/internal/ethzasl_brisk/brisk/include/brisk/internal/image-down-sampling.h:44:23: fatal error: emmintrin.h: No such file or directory
```

<br/><br/>
- My environment is : 
```shell
$ uname -a
Linux CAR01-CU03 4.4.38 #1 SMP PREEMPT Tue Oct 10 15:24:32 CST 2017 aarch64 aarch64 aarch64 GNU/Linux
```

fabianbl, 2017-12-18:
Hi @megamindhe . It seems that Neon instructions must not be specified for `armv8` processors (see ethz-asl/maplab#7), you should be fine by commenting out all occurrences of both `-mssse3` and/or `-mfpu=neon`. Within the `feature/port-to-arm` branch in ethz-asl/ethzasl_brisk, you can do this by modifying the `agast/cmake/export_flags.cmake` file to the following:

```cmake
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -DHAVE_OPENCV -std=c++0x")
set(HAS_NEON FALSE)
```

I hope this helps. Would be interesting to hear if this works out for now :-)

mfehr, 2018-01-12:
Closing this issue. Please follow the main [ARM issue](https://github.com/ethz-asl/maplab/issues/7) for updates on the progress of ARM support and for further questions. We will try to provide an experimental release candidate branch with ARM support within the next couple of days. Special thanks to @fabianbl and @Alabate.

closed 2018-01-12 by mfehr

fabianbl, 2018-01-24:
@megamindhe The `pre-release-january` branch in both the `maplab` and the `maplab_dependencies` repo adds build compatibility for ARM processors, see here: ethz-asl/maplab#7