ethz-asl/maplab #19: Recommanded devices

Alabate, 2017-12-20:
Hi,
I tried to use Rovioli with low cost devices (oCam 1MGN-U Global shutter camera and MPU-6050), and here is the best result I got for two simple loop in my building:

[image]

I guess that the main problem is my IMU which may be too cheap. So, I would like to know what devices do you use ? It could be nice to add a wiki page with recommanded devices that you or other people tried (+ imu calibration).

schneith, 2017-12-20:
Hi @Alabate 
We are working on an update (which we plan to release in the next days) that will simplify the integration of new devices. It will unify the noise definitions of ROVIO and maplab so that only one set of IMU noises need to be defined. This release will also include the values for the BMX055 IMU which will support the following devices:
- Realsense ZR300, and
- Tango tablet (yellowstone),
additionally to the currently supported ADIS16488 that is used by the Skybotix VISensor. With the Realsense we hope to support a (somewhat) publicly available sensor out-of-the-box.

To your questions:
The drift you are seeing could have multiple causes. The most important that come to my mind are:
1. time synchronization between IMU and camera
2. camera calibration
3. noise definitions for the IMU
4. IMU calibration

1) First, try to make sure that the time sync between IMU and cam is correct. You can use [kalibr](https://github.com/ethz-asl/kalibr) to estimate the delay. Unfortunately, there is no option in ROVIOLIto compensate for such a delay and you would either need to implement it or compensate for it externally.

2) Make sure that the camera calibration is accurate. If you use kalibr for calibration you will get a report at the end that let's you see the re-projection errors of the target observations. For usual setups you want them to be smaller than ~0.5px.

3) If you are using the default IMU noise values for the ADIS16488, you will need to obtain values for the MPU-6050 for a good estimation performance. Janosch Nikolic has written a nice article on how to do this here: https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model . At the moment you still need to provide two sets of noise values one for maplab and one for ROVIO. The values for ROVIO can just be the squares of the maplab values. As mentioned initially, the need for the second ROVIO noise file will soon be removed from the framework.

4) As the MPU-6050 is a low-cost device it might make sense to calibrate the "intrinsics" of the IMU (axis misalignment, scale factors, cross-sensitivity,...). Kalibr can do that for you: https://github.com/ethz-asl/kalibr/wiki/Multi-IMU-and-IMU-intrinsic-calibration . Unfortunately, maplab can not correct for these effects at the moment and similar to the time delay you would need to implement the correction directly into the estimator or compensate it externally before feeding the IMU data to ROVIOLI.

After a correct calibration of all sensor models, I would expect the MPU-6050 to perform well with ROVIOLI.

mfehr, 2018-01-12:
Closing this issue now. We added an [entry on the FAQ page](https://github.com/ethz-asl/maplab/wiki/FAQ#q-what-sensors-are-recommended-for-maplab).

closed 2018-01-12 by mfehr