ethz-asl/maplab #20: stereo ROVIO

zz1120122573, 2017-12-21:
I set  -DROVIO_NCAM=2 in cmakelist of rovio, but I cannot ensure it run as stereo. I can see the video using 
visualize_raw_images, but nothing when using visualize_disparity_maps. Do I need to modify other parameters? I use Euroc dataset.

zz1120122573, 2017-12-23:
I solve the problem. I delete the rovio file in build, and rebuild.

mfehr, 2018-01-03:
If I understand your question(s) correctly there are several issues here:
 * ROVIO has some issues in stereo mode, we don't recommend using it if you don't need it. However ROVIOLI allows you to run ROVIO in monocular mode but build the map based on all the cameras specified in the ncamera config file (e.g. stereo for the euroc case). That's what [this file](https://github.com/ethz-asl/maplab/blob/master/applications/rovioli/share/ncamera-euroc-stereo.yaml) is for.
 * the `visualize_*` commands are intended to show image-based resources that are attached to the map. Therefore you will see the raw camera images if you call `visualize_raw_images`, however there are no disparity images attached to the map by default, because there is no stereo matcher running or anything of that sort. You can run `res_stats` to list the number of resources per type that are attached to a map.
 * If you followed the dense reconstruction tutorial and didn't get any depth information, then something obviously went wrong and we would need a more structured and informative bug report to help you :)

closed 2018-01-03 by mfehr

zz1120122573, 2018-01-04:
Thanks for your reply.I can see depth map follow the dense reconstruction tutorial. What's the problem of ROVIO Stereo? I want to run ROVIOLI in stereo mode. Do you mean that ROVIOLI can only run in monocular mode? The map is VImap or dense reconstruction map? Thanks a lot.

mfehr, 2018-01-04:
I believe this was answered in your issue in [ROVIO](https://github.com/ethz-asl/rovio/issues/144).  ROVIOLI will always run the underlying ROVIO in monocular mode, but build a VIMap that includes cameras and landmarks from all cameras in the camer calibration yaml. You can verify this by calling ms in the console, it should list two cameras. There is no "dense reconstruction map" just VImaps with or without the necessary resources (images) to create a dense reconstruction. The results of any dense reconstruction algorithm can then be stored as either per-vertex resources (e.g. disparity maps, depth maps, ...) or additionally these resources can be fused in a TSDF grid to produce a per-mission-set resource. Does this help?

zz1120122573, 2018-01-06:
Thanks, it's really helpful. Does OKVIS has the same problem with ROVIO stereo?