ethz-asl/maplab #21: Missing minus sign in line 30 in unit3-param.cc?

BITVoyager, 2017-12-22:
Would the following line be negating the coefficients of the quaternion? I guess a minus sign was missing.

https://github.com/ethz-asl/maplab/blob/63ea0c8acb669904a711817b3cb74e314ecffec7/algorithms/ceres-error-terms/src/parameterization/unit3-param.cc#L30

NikolausDemmel, 2018-01-03:
I'm not very familiar with maplab, but it looks like there was a normalization to positive `w` previously, then it was later removed (along with the commented out asserts).

To be honest I'm not sure why you would want to normalize the quaternion positive `w` for Jacobian computation in the first place (if someone thinks otherwise, please enlighten me). Without looking too much at the context, I think it should be fine as is, but the code could be cleaned up to remove `quat_x_copy` altogether, I guess.

mfehr, 2018-01-04:
@BITVoyager thanks for the report, this does indeed look fishy, we will look into it.

schneith, 2018-01-18:
Looks like a bug, it should indeed have a minus. Thanks for reporting.