ethz-asl/maplab #22: Can ORB feature work with Maplab?

aipiano, 2017-12-22:
How can I use maplab to build an orb-feature map?

mfehr, 2018-01-08:
Hi @aipiano, our data structures support binary descriptors in general (because we already support Brisk and FREAK), but it will take some additional effort to integrate ORB, especially in the feature-detection and also in the loop closure logic (e.g. training a new projection matrix for the inverted multi-index). If you are interested in pursuing this, please let us know and we might be able to help you with some pointers on where to change things.

dshwtc, 2018-04-17:
Hi @mfehr , I'm planning to integrate ORB into maplab, and can you give some pointers about this change?

dymczykm, 2018-04-17:
Hi @dshwtc,
the ROVIOLI module responsible for feature tracking and description is here:
https://github.com/ethz-asl/maplab/blob/master/applications/rovioli/include/rovioli/feature-tracking.h
As you can see, this code depends on the feature-tracking package:
https://github.com/ethz-asl/maplab/tree/master/algorithms/feature-tracking/include/feature-tracking
You would basically need to integrate your new descriptor there. The ORB detector is already supported.

Another task is to adapt the loopclosure to support the ORB features. The easiest would be to use the kd-tree engine (see the flag in https://github.com/ethz-asl/maplab/blob/master/algorithms/loopclosure/matching-based-loopclosure/src/detector-settings.cc) - then you would just need to train an appriopriate projection matrix and introduce another feature type (see https://github.com/ethz-asl/maplab/blob/master/algorithms/loopclosure/loopclosure-common/src/flags.cc for a flag switching between freak and brisk).

arszero, 2018-12-04:
Hi @dymczykm, thank you for the detailed answer and it really helpful. I am also trying to integrate other descriptors like ORB and Brisk (v1, 512bit size). But I am not sure what is the appropriate way to get the new projection matrix. Is it possible to also provide some instructions on how to train the projection matrix ?

EmreOzkose, 2019-12-15:
@mfehr Could you integrate ORB-slam features into maplab? If you did, can you share the script if it won't be a problem for you?

EmreOzkose, 2019-12-15:
Hi! @mfehr Could you integrate ORB-slam features into maplab? If you did, can you share the script if it won't be a problem for you?

mfehr, 2019-12-15:
@EmreOzkose nope.