ethz-asl/maplab #23: Doxygen failed

qinyusen, 2017-12-27:
Actions as follow:
1. cd src/maplab/maplab
2. doxygen-catkin-filegen -h maplab
3. change some options in Doxyfile.in 
4. add add_doxygen(REQUIRED) in CMakeList.txt
5. build as readme.

But got nothing in devel/share/maplab/doc 

So, I think there must be something wrong with my step, but after I tried to reproduce the similar steps into different module's folders CMakeList.txt, still shown nothing.

That may not a Issue, but I got no way out...

Thanks if could give any tips.  And Feel free to close this issue if you are not interested in reply.

Yusen QIN

mfehr, 2018-01-08:
Hi @qinyusen 
I'm afraid we don't have a lot of experience with doxygen in our team and maplab doesn't really use the doxygen comment style. However having a doxygen page would definitely be helpful, even without the comments, so if you are pursuing this we would of course appreciate it if you share the results with us.

The only thing I could think of is that maplab is a metapackage and that's why it's not working. Sorry for not being more helpful.

qinyusen, 2018-01-10:
Hi, @mfehr 

It's ok,  I will look forward to try to make it works. And will ask for a solution.

Since doxygen could simply show how funtions called  even no doxygen style commit , and also for maplab such a modularization project, 

And as a engineer in Segway Robotics, I just curious on how your team works for such a big project and make the debug? 

BR,
Yusen

dymczykm, 2018-01-12:
@qinyusen 
To navigate through the code, we mostly rely on our IDEs (e.g. Eclipse) that help to understand types and function signatures. We also try to name function/parameters so the code is as self-explanatory as possible.

For the debugging, we mostly rely on Google Logging Library by adding checks whenever possible (so you know where your code crashed -> it's easy to identify the bug). Additionally, unit tests and a continuous integration server helps to detect errors when changing the codebase.

closed 2018-08-15 by schneith