ethz-asl/maplab #25: build error

dan9thsense, 2017-12-29:
Using Ubuntu 16.04, ROS Kinetic, with opencv 3.2 already installed.

Getting this error while it is trying to build its own opencv:
fatal error: LAPACKE_H_PATH-NOTFOUND/lapacke.h: No such file or directory

ZacharyTaylor, 2018-01-03:
You need to install lapack, you can do this via:
sudo apt install liblapack-dev

maplab has quite a few dependencies, for the full list of packages you need installed see https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu

dan9thsense, 2018-01-03:
I already had liblapack-dev installed and already went through the wiki instructions, including dependencies, before posting here.
Here is a summary of what I have found so far abouth is error:
http://answers.opencv.org/question/121651/fata-error-lapacke_h_path-notfound-when-building-opencv-32/

Other ideas?

mfehr, 2018-01-04:
Hi @dan9thsense 
Could you run `catkin config` in your catkin ws and share it with us?

dan9thsense, 2018-01-04:
Sure, here you go:
Profile:                     default
Extending:          [cached] /opt/ros/kinetic
Workspace:                   /home/dbarry/catkin_ws
-----------------------------------------------------------
Source Space:       [exists] /home/dbarry/catkin_ws/src
Log Space:          [exists] /home/dbarry/catkin_ws/logs
Build Space:        [exists] /home/dbarry/catkin_ws/build
Devel Space:        [exists] /home/dbarry/catkin_ws/devel
Install Space:      [unused] /home/dbarry/catkin_ws/install
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.

eggerk, 2018-01-05:
Maplab doesn't work with a linked devel space layout. You need to switch to a merged layout. See this FAQ entry with instructions on how to switch: https://github.com/ethz-asl/maplab/wiki/FAQ#q-compilation-failed-because-protoc-was-not-found. Could you retry again with this setting?

Note that we also recommend to set the build type to release to ensure good performance when running maplab. You can do this by issuing the following command inside your catkin workspace:
```
catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release
```

dan9thsense, 2018-01-05:
Good news, that was the problem, it builds now. Thanks for the help.
 For future reference, here is the current catkin config:
```
-----------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/kinetic
Workspace:                   /home/dbarry/catkin_ws
-----------------------------------------------------------
Source Space:       [exists] /home/dbarry/catkin_ws/src
Log Space:          [exists] /home/dbarry/catkin_ws/logs
Build Space:        [exists] /home/dbarry/catkin_ws/build
Devel Space:        [exists] /home/dbarry/catkin_ws/devel
Install Space:      [unused] /home/dbarry/catkin_ws/install
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------
```

dan9thsense, 2018-01-06:
I don't know how to mark "answered," so I will just close it.

closed 2018-01-06 by dan9thsense