ethz-asl/maplab #26: Plotter must be non NULL

fredlllll, 2018-01-03:
so i followed the instructions to install and build maplab on ubuntu 16.04

then i tried to follow the "use basic console" instructions and got this error

```
F0103 14:11:54.528542 22025 console-plugin-base-with-plotter.h:23] Check failed: 'plotter' Must be non NULL
*** Check failure stack trace: ***
*** Aborted at 1514988714 (unix time) try "date -d @1514988714" if you are using GNU date ***
PC: @     0x7f439bbdf428 gsignal
*** SIGABRT (@0x3e800005609) received by PID 22025 (TID 0x7f439cd26b00) from PID 22025; stack trace: ***
    @     0x7f439bbdf4b0 (unknown)
    @     0x7f439bbdf428 gsignal
    @     0x7f439bbe102a abort
    @           0x402973 common::glogFailFunctionWithConsoleRecovery()
    @     0x7f439c72326d google::LogMessage::Fail()
    @     0x7f439c725083 google::LogMessage::SendToLog()
    @     0x7f439c722dfb google::LogMessage::Flush()
    @     0x7f439c725a6e google::LogMessageFatal::~LogMessageFatal()
    @     0x7f4386063c85 google::CheckNotNull<>()
    @     0x7f4386052475 vi_map::VIMapBasicPlugin::VIMapBasicPlugin()
    @     0x7f4386054c75 createConsolePlugin
    @     0x7f439c511e47 maplab::MapLabConsole::discoverAndInstallPlugins()
    @     0x7f439c5127c7 maplab::MapLabConsole::MapLabConsole()
    @           0x402694 main
    @     0x7f439bbca830 __libc_start_main
    @           0x402739 _start
runmaplab.sh: line 2: 22025 Aborted                 (core dumped) rosrun maplab_console maplab_console --ros_free
```

i even ran the build again and it said something about skipping 14 things, but no reason why. there were no failures

any idea what went wrong?

mfehr, 2018-01-03:
Hi @fredlllll 
Did you run maplab in ros-free mode? If that's the case, that would explain the issue. It doesn't really work at the moment. The ros-free mode was introduced in the early days of maplab, but never really used, so not all plugins respect the possibility that there is no visualization available (no plotter, as it is ROS-based).

fredlllll, 2018-01-03:
hi,
yes i used the ros_free option, cause i didnt really know if i need it. the wiki doesnt mention that its broken.

i would like to use maplab to convert a series of images with gps info into a textured 3d model. is that possible? i dont have a robot or anything, just the image series with gps in its exif data

mfehr, 2018-01-04:
We do list it as a known issue: https://github.com/ethz-asl/maplab/wiki/Known-Issues

Maplab focuses on Visual-Inertial mapping, unless you have IMU measurements as well you won't be able to obtain good maps without additional effort, like wrapping a visual only pipeline (or converting it's output), like ORB-SLAM. I recommend to look into an offline  SFM pipeline. There is a nice list of available frameworks by Pierre Moulon here: https://github.com/openMVG/awesome_3DReconstruction_list

closed 2018-01-04 by mfehr