ethz-asl/maplab #30: Error when Running ROVIOLI in VIO mode with Building from a rostopic

zhuzunjie17, 2018-01-09:
system -- ubuntu 14.04.5

when i Running ROVIOLI in VIO mode: building from a rostopic, i follow the command:
`source ~/maplab_ws/devel/setup.bash`
`roscore&`
`rosrun rovioli tutorial_euroc_live save_folder`
then i got the error show bellow:
```
zhuzunjie@ubuntu14045:~/maplab_ws$ rosrun rovioli tutorial_euroc_live save_folder
I0109 16:56:04.857854 128866 rovio-factory.cc:169] Loading ROVIO configuration template: /home/zhuzunjie/maplab_ws/src/maplab/applications/rovioli/share//rovio_default_config.info
Set distortion parameters (Radtan) to: k1(-0.283408), k2(0.0739591), k3(0), p1(0.000193595), p2(1.76187e-05)
*** Aborted at 1515488164 (unix time) try "date -d @1515488164" if you are using GNU date ***
PC: @     0x7fc35e7f108f std::string::_Rep::_M_dispose()
*** SIGSEGV (@0xfffffffffffffff8) received by PID 128866 (TID 0x7fc3611beb80) from PID 18446744073709551608; stack trace: ***
    @     0x7fc35eeba330 (unknown)
    @     0x7fc35e7f108f std::string::_Rep::_M_dispose()
    @     0x7fc3603e7621 aslam::NCamera::~NCamera()
    @     0x7fc35ec3ca9f rovioli::RovioFlow::RovioFlow()
    @     0x7fc35ec42019 rovioli::RovioliNode::RovioliNode()
    @           0x40bf7f main
    @     0x7fc35e16df45 __libc_start_main
    @           0x40c83d (unknown)
/home/zhuzunjie/maplab_ws/src/maplab/applications/rovioli/scripts/tutorials/tutorial_euroc_live: line 20: 128866 Segmentation fault      (core dumped) rosrun rovioli rovioli --alsologtostderr=1 --v=2 --ncamera_calibration=$NCAMERA_CALIBRATION --imu_parameters_maplab=$IMU_PARAMETERS_MAPLAB --imu_parameters_rovio=$IMU_PARAMETERS_ROVIO --datasource_type="rostopic" --save_map_folder="$LOCALIZATION_MAP_OUTPUT" --map_builder_save_image_as_resources=false --optimize_map_to_localization_map=false $REST
```
what i supposed to do?
thanks in advance!

dymczykm, 2018-01-09:
Hi, 
thanks for your bug report. That indeed looks like a bug but we have never encountered this issue before.

Could you please:
* report your compiler version (``gcc --version``)
* clean your catkin workspace  (``catkin clean``) and build from scratch and check again
* check if the issue also appears when runing ROVIO from rosbag?

zhuzunjie17, 2018-01-10:
Thank you! I figure it out by following your advice. It is due to the higher version of compiler!

Honestly, the memory back to my mind after following your advice. The first time i build the project is in `gcc 4.8` . After that, i build other project (maybe intel realsense ZR300, cannot remember) which need gcc version higer than 4.9, so i switch my gcc to 5.4.1 and then occurred this issue.
After i swhich my compiler back to 4.8, everything works fine!

Note: the same issue occurred when running ROVIO from rosbag in gcc 5.4.1.

dymczykm, 2018-01-11:
@zhuzunjie17 thanks for your report! At least we know what was the cause.

closed 2018-01-12 by mfehr