ethz-asl/maplab #31: No ros version

zz1120122573, 2018-01-09:
Hi, will you release a no ros version for maplab? Is it possible to do this work for one person in two weeks?

dymczykm, 2018-01-09:
Hi, thanks for your question. 

Vast majority of maplab code (including algorithms) does not depend on ROS at all. ROS is mostly used:
* for visualization with RViz
* for reading ROS topics with input data and publishing estimator output in ROVIOLI

I believe it's definitely possible to do this work within 2 weeks.

zz1120122573, 2018-01-10:
Thanks for your reply. I'm afraid I have to spend more than two weeks to understand the code. Is there any tutorials can help me to understand the code structure and data flow? Can I directly use the vocabulary tree? It's ros vocabulary tree.

dymczykm, 2018-01-11:
@zz1120122573 
The only documentation is provided withing the project wiki.

Why would you like to use ROS vocabulary tree?

zz1120122573, 2018-01-19:
@dymczykm I know I don't need to retrain a vocabulary tree, but I thought may be the library is needed for KNN search. I'm struggling to recompile the program. I want to move maplab directly to Windows. However, I don't know a lot about cmake. I may give up. Thanks.

dymczykm, 2018-01-23:
@zz1120122573 I think moving maplab to Windows might be pretty challenging because of many dependencies. I would really recommend to consider a virtual machine if you would like to experiment with the framework.

closed 2018-01-30 by dymczykm