ethz-asl/maplab #32: How to use realsenseZR300 with vio rostopic mode

zhuzunjie17, 2018-01-10:
I am really a new to ROS, the thing is that i want to use realsenseZR300 with vio rostopic mode.
all i have done is :    change the label "`cam0`" to "`camera/color`"(realsenseZR300 color camera topic is /camera/color/image_raw), i change the other .yaml file as well.

but it feedback:
```
W0110 15:19:20.265130 90386 imu-camera-synchronizer.cc:103] No IMU messages have been received in the last 5 seconds. Check for measurement drops or if the topic is properly set in the maplab IMU configuration file
W0110 15:19:20.265285 90386 imu-camera-synchronizer.cc:111] No camera messages have been received in the last 5 seconds. Check for measurement drops or if the topic is properly set in the camera configuration file
```
is anywhere else i should change? 

I know i should learn ROS first, then ask. But what i want to do is only do some experimemts, i.e. give maplab some input and see the results. I don't know how much cost will it be to learn ROS prior and whether its worth.

I sincerely wish someone could show me the path!
Thanks in advance.

dymczykm, 2018-01-10:
@zhuzunjie17 could you run the ZR300 driver and then call ``rostopic list`` in another terminal window and paste the output here?

zhuzunjie17, 2018-01-10:
@dymczykm 
here it is:
```
/cam0/image_raw
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/depth/camera_info
/camera/depth/image_raw
/camera/depth/image_raw/compressed
/camera/depth/image_raw/compressed/parameter_descriptions
/camera/depth/image_raw/compressed/parameter_updates
/camera/depth/image_raw/compressedDepth
/camera/depth/image_raw/compressedDepth/parameter_descriptions
/camera/depth/image_raw/compressedDepth/parameter_updates
/camera/depth/image_raw/theora
/camera/depth/image_raw/theora/parameter_descriptions
/camera/depth/image_raw/theora/parameter_updates
/camera/points
/clicked_point
/imu0
/initialpose
/maplab_rovio/T_G_I
/maplab_rovio/T_G_M
/maplab_rovio/T_M_I
/maplab_rovio/bias_acc
/maplab_rovio/bias_gyro
/maplab_rovio/velocity_I
/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static
```

zhuzunjie17, 2018-01-10:
@dymczykm 
sorry, there maybe other rosbag topics in it

dymczykm, 2018-01-10:
@zhuzunjie17 it seems that your camera topic is ``/cam0/image_raw``, are you sure  this is a correct bag?

zhuzunjie17, 2018-01-10:
@dymczykm 
here again:
```
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/depth/camera_info
/camera/depth/image_raw
/camera/depth/image_raw/compressed
/camera/depth/image_raw/compressed/parameter_descriptions
/camera/depth/image_raw/compressed/parameter_updates
/camera/depth/image_raw/compressedDepth
/camera/depth/image_raw/compressedDepth/parameter_descriptions
/camera/depth/image_raw/compressedDepth/parameter_updates
/camera/depth/image_raw/theora
/camera/depth/image_raw/theora/parameter_descriptions
/camera/depth/image_raw/theora/parameter_updates
/camera/points
/clicked_point
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static
```

dymczykm, 2018-01-10:
@zhuzunjie17 here, on the other hand, I don't see any IMU topic which is necessary to start the estimator

zhuzunjie17, 2018-01-10:
@dymczykm 
i run ZR300 in another mode, and get informations below:
```
/camera/accel/imu_info
/camera/accel/sample
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/depth/camera_info
/camera/depth/image_raw
/camera/depth/image_raw/compressed
/camera/depth/image_raw/compressed/parameter_descriptions
/camera/depth/image_raw/compressed/parameter_updates
/camera/depth/image_raw/compressedDepth
/camera/depth/image_raw/compressedDepth/parameter_descriptions
/camera/depth/image_raw/compressedDepth/parameter_updates
/camera/depth/image_raw/theora
/camera/depth/image_raw/theora/parameter_descriptions
/camera/depth/image_raw/theora/parameter_updates
/camera/extrinsics/fisheye2depth
/camera/extrinsics/fisheye2imu
/camera/fisheye/camera_info
/camera/fisheye/image_raw
/camera/fisheye/image_raw/compressed
/camera/fisheye/image_raw/compressed/parameter_descriptions
/camera/fisheye/image_raw/compressed/parameter_updates
/camera/fisheye/image_raw/compressedDepth
/camera/fisheye/image_raw/compressedDepth/parameter_descriptions
/camera/fisheye/image_raw/compressedDepth/parameter_updates
/camera/fisheye/image_raw/theora
/camera/fisheye/image_raw/theora/parameter_descriptions
/camera/fisheye/image_raw/theora/parameter_updates
/camera/gyro/imu_info
/camera/gyro/sample
/camera/points
/clicked_point
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static

```

dymczykm, 2018-01-10:
@zhuzunjie17 still, I don't see any IMU information, which ZR300 driver are you using?

zhuzunjie17, 2018-01-10:
@dymczykm 
i dont know where to see the driver. is this one?: 1.12.1 , or firmware you mean?
my system is Ubuntu 14.04.5 ,  kernel is : 4.4

zhuzunjie17, 2018-01-10:
@dymczykm 
the information of my ZR300 is as below:
```
Device 0 - Intel RealSense ZR300:
 Serial number: 3511804114
 Firmware version: 2.0.71.28
 USB Port ID: 4-4-1-2
 Adapter Board Firmware version: 1.29.0.0
 Motion Module Firmware version: 1.25.0.0
 Camera info: 
    DEVICE_NAME         :       Intel RealSense ZR300
    DEVICE_SERIAL_NUMBER:       3511804114
    CAMERA_FIRMWARE_VERSION:    2.0.71.28
    ADAPTER_BOARD_FIRMWARE_VERSION:     1.29.0.0
    MOTION_MODULE_FIRMWARE_VERSION:     1.25.0.0
    CAMERA_TYPE         :       PRQ-Ready
    OEM_ID              :       OEM None
    ISP_FW_VERSION      :       0xd942
    CONTENT_VERSION     :       12
    MODULE_VERSION      :       4.4.0.111
    IMAGER_MODEL_NUMBER :       33
    CALIBRATION_DATE    :       2014-07-04 08:18:35 UTC
    EMITTER_TYPE        :       Laser Driver 4.1
    FOCUS_VALUE         :       0
    LENS_TYPE           :       Newmax 58.9 x 45.9 degs in VGA
    3RD_LENS_TYPE       :       AEOT
    LENS_COATING__TYPE  :       Visible-light block / IR pass 43 nm width
    3RD_LENS_COATING_TYPE:      IR coating
    NOMINAL_BASELINE    :       70 mm
    3RD_NOMINAL_BASELINE:       58 mm

```

dymczykm, 2018-01-10:
@zhuzunjie17 which realsense ROS node are you using?

zhuzunjie17, 2018-01-10:
@dymczykm 
sorry, i dont exactly know what the node is, and where can i get the information?

zhuzunjie17, 2018-01-10:
@dymczykm 
if you meam the branches or tags of `realsense_samples_ros` , then i dont remember exactly, maybe `kinetic-devel`

ZacharyTaylor, 2018-01-10:
There are several different repos that can be used to run the zr300 in ros and not all of them give the same output. For example the official realsense ros drivers (https://github.com/intel-ros/realsense) have some time synchronization issues and the master branch no longer supports some sensors. 

We have our own realsense ros driver in this repo https://github.com/ethz-asl/maplab_realsense

zhuzunjie17, 2018-01-10:
@ZacharyTaylor @dymczykm 
Thanks both of you, it really helps me a lot!!! Now I can run maplab with realsense!

dymczykm, 2018-01-11:
@zhuzunjie17 we're glad it worked for you :+1:

mfehr, 2018-01-12:
Thanks for the report, we've added an [entry in the FAQ](https://github.com/ethz-asl/maplab/wiki/FAQ#q-how-can-i-use-the-intel-realsense-zr300) and a [wiki page](https://github.com/ethz-asl/maplab/wiki/Intel-Realsense-ZR300) with example calibration for the ZR300.

closed 2018-01-12 by mfehr