ethz-asl/maplab #37: How to rewrite IMU optimization component?

HTLife, 2018-01-23:
# VIO with rotation only
The [Running ROVIOLI in VIO mode](https://github.com/ethz-asl/maplab/wiki/Running-ROVIOLI-in-VIO-mode) optimize visual odometry by "rotation" and "translation" data from IMU.  However, I want to modify this part to use "**rotation**" only to optimize visual odometry.

Could you give me some suggestion that how can I do this (how can I modify the code)?

# Incorporating GPS signal
In addition, I also want to incorporate GPS into maplab, I aware that **"How to integrate a new sensor: coming soon (IMU-Sigmas - Janosch's tutorial)"** might be a relative topic (although it's not yet available).  How can I do? (same process with the question above?)

dymczykm, 2018-01-23:
Hi @HTLife,
thanks for your questions.

1. I assume you only have gyroscope measurements available, is that correct? In this case you would need to modify Inertial Error Term (https://github.com/ethz-asl/maplab/blob/master/algorithms/ceres-error-terms/include/ceres-error-terms/inertial-error-term.h). The idea of the error term is to compare the relative pose between two frames with the values from the integrated IMU measurements. You would also need to write a gyro-only IMU integrator (see https://github.com/ethz-asl/maplab/tree/master/algorithms/imu-integrator-rk4)

2. Firstly, I would like to say that we are planning to introduce GPS support at some point. Basically, you're missing a few components:
* Feeding GPS data to ROVIOLI (subscribing to the ROS topics etc.)
* Optional: Feeding GPS estimates to ROVIO so that they are also used for odometry estimates
* Passing GPS readings to VI-map that can be read by maplab
* Writing optimization error terms that use the GPS data stored in the VI-map (we might help you with this one)

HTLife, 2018-01-24:
@dymczykm Thanks for the reply with details.

1. Yes, I'm trying to use gyroscope alone to optimize the relative pose.  I can now get what error-term file do.  On the other hand, what do IMU integrator do?
2. For the GPS optimization error terms, do you have any reference paper recommended?

kauevestena, 2018-01-28:
So, and what about the "Zurich old town dataset ", the map are not Georeferenced? Or this was done by other means? 
I really want to obtain georeferenced maps by using maplab...

dymczykm, 2018-01-30:
@HTLife 
1. IMU integrator takes the IMU measurements that were recorded between a pair of consecutive keyframes, uses the current bias estimates (stored in vertices) and forward propagates the IMU. It also returns a covariance estimate.
2. I don't have anything at hand, but you can just treat your GPS measurement as an absolute position 3-DoF measurement and integrate it this way.

@kauevestena 
1. They are indeed not georeferenced. It was only a visual-inertial dataset. The alignment for the visualization purposes was done manually.
2. I can't really promise you any date, but we are working on adding the GPS measurements to maplab.

kauevestena, 2018-02-02:
@dymczykm Thank for your answers.

1 . So, for the moment, you suggest any tool for georeference maplab maps?

2 . So, please, let me know if there is any kind of task that may I can help the maplab team, in order to implement the GPS/GNSS funcionality. I will be very pleased in giving any contribution. Do you plan to deal with tightly coupling raw satellite data, or losely coupling positional data?

closed 2018-02-04 by HTLife