ethz-asl/maplab #39: Difference of two imu parameters?

xukuanHIT, 2018-01-26:
Hi, I just want to know why there have two imu parameters files for euroc dataset? They have different meaning?

---

Before submitting this issue, please consider the following **checklist**:

 - Consult the [FAQ](https://github.com/ethz-asl/maplab/wiki/FAQ) and [Known Issues](https://github.com/ethz-asl/maplab/wiki/Known-Issues) pages.
 - Make sure maplab was installed using the proper [installation instructions](https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu)
 - For bugs and build problems please add (if applicable):
   - Version of:
     - Operating System (32/64bit, 14.04/16.04/...)
     - Eigen
     - ROS
   - A backtrace (if maplab crashed)
   - Run `catkin config` in your workspace and post it here.
   - Details about the commands/scripts used.
   - Details about special parameter values or manual changes to the system.
   - Pictures/screen recordings
 - For general questions:
   - Please elaborate, we are no mind readers :)

dymczykm, 2018-01-30:
Hi @xukuanHIT,

right now, in the ``master`` branch, we have separate sets of parameters:
1. One for ROVIO/ROVIOLI, i.e. the visual-inertial odometry estimator.
2. One for maplab console, used for offline algorithms (mostly ``optvi`` command, so our visual-inertial least squares optimization).

In the upcoming update of maplab, these paremeter sets will get unified.

xukuanHIT, 2018-01-30:
@dymczykm 
Thank you for reply.  I can get one set of parameters from user manual of imu now. How can I get the another? What is the relation between them?

pxlong, 2018-03-06:
I have the same issue. How to decide the imu parameters in imu_parameter_maplab and imu_parameters_rovio?

mfehr, 2018-03-06:
Hi @xukuanHIT @pxlong, basically the two sets of parameters originated from ROVIO and maplab using two different conventions for the IMU parameters. We have unified them in our upcoming release of maplab. You can simply try to do what we did there:

 * take the square of the maplab values for ROVIO.

ROVIO is using the variance, i.e. the square of the standard deviation (which is what maplab is using). However, if you look at the values we provided for the VI-Sensor, you'll notice that that doesn't really hold there, because values for the VI-Sensor have been tuned independently for maplab and ROVIO by different people in our lab to a point where they were satisfied with the performance of the estimator. The point being that you should start with the square of the maplab values, but if it doesn't work well you will probably need to tune them a bit yourselves.

closed 2018-03-08 by xukuanHIT