ethz-asl/maplab #41: No Unified Projection Support?

cloitre, 2018-01-30:
Hello,

I was about to test Rovioli in VIO mode using my dataset which consists of fisheye cameras and I have successfully translated my Kalibr camchain into a Rovioli ncamera yaml file.
I however receive the following error when I launch Rovioli:

F0130 16:38:04.192309 26766 rovio-factory.cc:52] Check failed: aslam_camera.getType() == aslam::Camera::Type::kPinhole (kUnifiedProjection vs. kPinhole) Only pinhole models supported.
*** Check failure stack trace: ***
    @     0x7f2e0e53f26d  google::LogMessage::Fail()
    @     0x7f2e0e541083  google::LogMessage::SendToLog()
    @     0x7f2e0e53edfb  google::LogMessage::Flush()
    @     0x7f2e0e541a6e  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f2e0db10013  rovioli::constructAndConfigureRovio()
    @     0x7f2e0dc23fed  rovioli::RovioFlow::RovioFlow()
    @     0x7f2e0dc2aa02  rovioli::RovioliNode::RovioliNode()
    @           0x40b632  main
    @     0x7f2e0d0a2830  __libc_start_main
    @           0x40bf09  _start

Did I miss a parameter to let Rovioli know I am using the Unified Projection model? Or is it not supported at all for now and Rovioli can only accept Pinhole models?

Regards,

Audren

dymczykm, 2018-01-31:
Hi @cloitre,
indeed, currently both ROVIO and some components of maplab (e.g. loopclosure PnP solver) don't support unified projection model. 

Hint: we have successfully used fisheye lenses with pinhole + epipolar distortion model, maybe you could give it a try?

cloitre, 2018-01-31:
OK, thanks @dymczykm.

I will keep an eye out to see if this feature gets added.

Regards,

Audren

closed 2018-02-02 by dymczykm