ethz-asl/maplab #45: IMU update rate

HTLife, 2018-02-19:
I modified the IMU update rate to be same with the image frame rate(timestamp also same with each other).  And create a rosbag for maplab.
Will "ROVIOLI in VIO mode" run correctly under this rate?  If not, how can I correct this issue?

I got some related error:
```
-- Filter: Initialized at t = 1413393212.28
W0219 16:41:30.055050 28081 imu-measurements-buffer.cc:165] Timeout reached while trying to get the requested IMU data. Requested range: 1413393212255760500 to 1413393212355760500.
W0219 16:41:30.055414 28081 imu-measurements-buffer.cc:171] The relevant IMU data will never be available. Either the buffer is too small or a sync issue occurred.
E0219 16:41:30.055459 28081 imu-camera-synchronizer.cc:173] Camera/IMU data out-of-order. This might be okay during initialization.
I0219 16:41:30.208765 28073 OutlierDetection.hpp:47] Detected outlier - mahalanobis (d / threshold):12.5689 > 9.21 - count : 1.
```

dymczykm, 2018-02-19:
@HTLife I don't think that the estimator will work reliably with an IMU rate equal to the image rate (which I assume to be around 30Hz). What's your framerate?

HTLife, 2018-02-19:
@dymczykm Yes, my image rate is around 30Hz.

dymczykm, 2018-02-19:
@HTLife what do you mean by saying "My IMU data is estimated from the image of EuRoC dataset."?

closed 2018-02-23 by HTLife

Jeoker, 2018-04-12:
Dear @HTLife ,
I am facing the same problem as you were. Did you figure out how to solve this problem? we are setting IMU at 200 Hz and image at 25 Hz

HTLife, 2018-04-13:
@Jeoker Do interpolation on low data rate signal.

Jeoker, 2018-04-14:
Thank you. I will check with that