ethz-asl/maplab #46: JPL convention of Quaternions

hitcm, 2018-02-23:
Thanks for your hard work on maplab, which give me many great ideas. I have little experiences on JPL convention of Quaternions. So questions confuse  me  a lot. 

**1.Convert active Hamiltonian rotation (minkindr, Eigen) to passive JPL**

I read from Joan Sola's tech report

http://www.iri.upc.edu/people/jsola/JoanSola/objectes/notes/kinematics.pdf

where he said both Hamiton and JPL are passive on page 36

**2. Quaternion inverse or not?**

https://github.com/ethz-asl/maplab/blob/master/algorithms/map-optimization/src/optimization-state-buffer.cc#L196  has no inverse

https://github.com/ethz-asl/maplab/blob/master/algorithms/map-optimization/src/optimization-state-buffer.cc#L218 has an inverse

And what's the difference?

**3.May be a typo?**

https://github.com/ethz-asl/maplab/blob/master/algorithms/ceres-error-terms/include/ceres-error-terms/visual-error-term-inl.h#L186

Do we have a minus here?

NikolausDemmel, 2018-02-23:
I haven't checked out your specific questions, but just FYI, here is a recent discussion on JPL vs Hamilton that also contains pointers to other discussions and blog posts: https://github.com/ethz-asl/kalibr/issues/161#issuecomment-362712790. Maybe that helps.

schneith, 2018-02-23:
You can find a nice discussion of the quaternion convention mess here: https://arxiv.org/pdf/1801.07478.pdf

Maplab uses the passive-Hamilton convention (as used in minkindr and Eigen) with one exception. For historic reasons parts of the optimization uses the JPL convention. The code that you have linked perform these conversions.

We'll try to give a more detailed answer soon.

NikolausDemmel, 2018-02-23:
@schneith, thanks for the link. This looks very useful, I have to check it out asap.

hitcm, 2018-02-24:
@schneith Great thanks.That's really helpful. I am trying to do the optimization using the Hamilton convention. Hope it works soon.

closed 2018-02-24 by hitcm