ethz-asl/maplab #47: rovioli node can not run on the desktop without the desktop environment

BryceeeeeChen, 2018-03-01:
Hi, I'm trying to rosrun rovioli on an Ubuntu server platform and use another computer to operate it. But the node will die after "cannot open display" error. I did not experience this kind of issue when I used the desktop version Ubuntu. And I have tried set MAKE_SCENE and DMAKE_SCENE to 0 in the maplab_dependencies/internal/maplab_rovio/CMakeLists.txt, which doesn't solve this problem. Could anyone help me with this? Thanks! 
[image]

eggerk, 2018-03-01:
Hi @BryceeeeeChen,

you can set the flag `--rovio_enable_frame_visualization=false` when running rovioli to disable the graphical output.

BryceeeeeChen, 2018-03-01:
Thank you @eggerk ! Problem solved.

closed 2018-03-01 by BryceeeeeChen