ethz-asl/maplab #48: ROVIOLI ccrashing at start

MetaI, 2018-03-05:
Hi dear Maplab team;

I calibrated a Lenovo Phab 2 Pro Tango device with kalibr, but after recording a rosbag and use rovioli, or trying to execute rovioli from a rostopic I got error every time. It seem start preview the image of the camera and features (I'm using the fisheye camera of the device) for a second but after that it crashes with the following log (I hightlighted in  bold the error message ). I'm using usb thetering for improving the bandwitch and  I got 30hz in camera and 200Hz in IMU. Could you see any clue here? Thanks in advance 

`E0305 10:09:21.436723 32714 yaml-serialization.h:43] Key "id" does not exist
W0305 10:09:21.437057 32714 sensor.cc:49] Unable to find an ID field. Generating a new random id.
I0305 10:09:21.475286 32714 rovio-factory.cc:178] Loading ROVIO configuration template: /home/innovae/Escritorio/maplab_enero_2018_ws/src/maplab/applications/rovioli/share//rovio_default_config.info
Set distortion parameters (fov) to: w(0.926948)
I0305 10:09:21.628355 32714 rovio-localization-handler.cc:92] Localization mode: structure constraints.
Registered filter state update callback.
I0305 10:09:21.700196 32714 map-builder-flow.cc:31] Set VIMap folder to: save_folder_32
I0305 10:09:21.714289 32714 rovioli-node.cc:99] Starting data source...
I0305 10:09:21.714375 32714 rovioli-app.cc:140] 
Message delivery queues:
subscriber-node               queue-topic                   queue-id                      num elements                  
MapBuilderFlow                ROVIO_ESTIMATES               d6df..0000                    0                             
MapBuilderFlow                VIO_UPDATES                   45a8..0000                    0                             
RovioFlow                     LOCALIZATION_RESULT           d606..0000                    0                             
DataPublisherFlow             ROVIO_ESTIMATES               23e7..0000                    0                             
RovioFlow                     IMAGE_MEASUREMENTS            1cf3..0000                    0                             
FeatureTrackingFlow           ROVIO_ESTIMATES               a527..0000                    0                             
DataPublisherFlow             LOCALIZATION_RESULT           64d4..0000                    0                             
ImuCameraSynchronizerFlow     IMAGE_MEASUREMENTS            1f91..0000                    0                             
RovioFlow                     IMU_MEASUREMENTS              a3b6..0000                    0                             
MapBuilderFlow                LOCALIZATION_RESULT           cbe5..0000                    0                             
MapBuilderFlow                TRACKED_NFRAMES_AND_IMU       95d4..0000                    0                             
DataPublisherFlow             RAW_VIMAP                     9ca2..0000                    0                             
ImuCameraSynchronizerFlow     IMU_MEASUREMENTS              759b..0000                    0                             
FeatureTrackingFlow           SYNCED_NFRAMES_AND_IMU        e60a..0000                    0                             
-- Filter: Initializing using accel. measurement ...
-- Filter: Initialized at t = 1
[FilterBase::addUpdateMeas] Warning: included measurements at time 0.966738007 before safeTime 1

**W0305 10:09:21.951208 32732 rovio-flow.cc:148] ROVIO rejected image measurement. Latency is too large.
F0305 10:09:22.006186 32732 imu-measurements-buffer-inl.h:11] Check failed: timestamp_nanoseconds > last_value.timestamp (1520240962222000000 vs. 1520240962222000000) Timestamps not strictly increasing.**

*** Check failure stack trace: ***
F0305 10:09:22.006188 32734 FilterBase.hpp:34] Check failed: measMap_.count(t) == 0u (1 vs. 0) 
*** Check failure stack trace: ***
    @     0x7f8b728df26d  google::LogMessage::Fail()
    @     0x7f8b728df26d  google::LogMessage::Fail()
    @     0x7f8b728e1083  google::LogMessage::SendToLog()
    @     0x7f8b728e1083  google::LogMessage::SendToLog()
    @     0x7f8b728dedfb  google::LogMessage::Flush()
    @     0x7f8b728dedfb  google::LogMessage::Flush()
    @     0x7f8b728e1a6e  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f8b728e1a6e  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f8b71e59fc8  vio_common::ImuMeasurementBuffer::addMeasurements()
    @     0x7f8b71efb8ad  LWF::MeasurementTimeline<>::addMeas()
    @     0x7f8b71e588b3  rovioli::ImuCameraSynchronizer::addImuMeasurements()
    @     0x7f8b71f937a3  rovio::RovioInterfaceImpl<>::processImuUpdate()
    @     0x7f8b71fc38d1  _ZNSt17_Function_handlerIFvRKSt10shared_ptrIN3vio14ImuMeasurementEEEZN7rovioli25ImuCameraSynchronizerFlow19attachToMessageFlowEPN12message_flow11MessageFlowEEUlS5_E0_E9_M_invokeERKSt9_Any_dataS5_
    @     0x7f8b71fb2e82  _ZNSt17_Function_handlerIFvRKSt10shared_ptrIN3vio14ImuMeasurementEEEZN7rovioli9RovioFlow19attachToMessageFlowEPN12message_flow11MessageFlowEEUlRKS0_IKS2_EE_E9_M_invokeERKSt9_Any_dataS5_
    @     0x7f8b71fb67eb  message_flow::MessageDeliveryQueue<>::deliverOldestMessage()
    @           0x40c72c  std::_Function_handler<>::_M_invoke()
    @           0x40cbc9  std::__future_base::_State_baseV2::_M_do_set()
    @     0x7f8b71fb67eb  message_flow::MessageDeliveryQueue<>::deliverOldestMessage()
    @           0x40c72c  std::_Function_handler<>::_M_invoke()
    @           0x40cbc9  std::__future_base::_State_baseV2::_M_do_set()
    @     0x7f8b72244a99  __pthread_once_slow
    @           0x40ce43  _ZNSt17_Function_handlerIFvvEZN5aslam10ThreadPool14enqueueOrderedISt5_BindIFSt7_Mem_fnIMN12message_flow24MessageDeliveryQueueBaseEFvvEEPS7_EEJEEESt6futureINSt9result_ofIFT_DpT0_EE4typeEEmOSG_DpOSH_EUlvE_E9_M_invokeERKSt9_Any_data
    @     0x7f8b72244a99  __pthread_once_slow
    @           0x40ce43  _ZNSt17_Function_handlerIFvvEZN5aslam10ThreadPool14enqueueOrderedISt5_BindIFSt7_Mem_fnIMN12message_flow24MessageDeliveryQueueBaseEFvvEEPS7_EEJEEESt6futureINSt9result_ofIFT_DpT0_EE4typeEEmOSG_DpOSH_EUlvE_E9_M_invokeERKSt9_Any_data
    @     0x7f8b72d38638  aslam::ThreadPool::run()
    @     0x7f8b72d38638  aslam::ThreadPool::run()
    @     0x7f8b71a5bc80  (unknown)
    @     0x7f8b71a5bc80  (unknown)
    @     0x7f8b7223d6ba  start_thread
    @     0x7f8b7223d6ba  start_thread
    @     0x7f8b714ca41d  clone
/home/innovae/Escritorio/maplab_enero_2018_ws/src/maplab/applications/rovioli/share/rovioli_vio_lenovo_live: línea 18: 32714 Abortado                (`core' generado) rosrun rovioli rovioli --alsologtostderr=1 --v=2 --ncamera_calibration=$NCAMERA_CALIBRATION --imu_parameters_maplab=$IMU_PARAMETERS_MAPLAB --datasource_type="rostopic" --save_map_folder="$LOCALIZATION_MAP_OUTPUT" --map_builder_save_image_as_resources=false --optimize_map_to_localization_map=false $REST'

**TODO:** ADD YOUR QUESTION/PROBLEM DESCRIPTION HERE

---

Before submitting this issue, please consider the following **checklist**:

 - Consult the [FAQ](https://github.com/ethz-asl/maplab/wiki/FAQ) and [Known Issues](https://github.com/ethz-asl/maplab/wiki/Known-Issues) pages.
 - Make sure maplab was installed using the proper [installation instructions](https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu)
 - For bugs and build problems please add (if applicable):
   - Version of:
     - Operating System (32/64bit, 14.04/16.04/...)
     - Eigen
     - ROS
   - A backtrace (if maplab crashed)
   - Run `catkin config` in your workspace and post it here.
   - Details about the commands/scripts used.
   - Details about special parameter values or manual changes to the system.
   - Pictures/screen recordings
 - For general questions:
   - Please elaborate, we are no mind readers :)

dymczykm, 2018-03-05:
Hi @MetaI ,
thanks for your report. After a look over your output, it seems that you're getting a messages with the same timestamp in your Tango output:
```
F0305 10:09:22.006186 32732 imu-measurements-buffer-inl.h:11] Check failed: timestamp_nanoseconds > last_value.timestamp (1520240962222000000 vs. 1520240962222000000) Timestamps not strictly increasing.
```
IMU buffer complains as there are two messages with the same timestamp 1520240962222000000. One way would be to take a look at ``imu-measurements-buffer`` and just turn off the requirement for the timestamps to be strictly increasing. 

Does that help you?

MetaI, 2018-03-05:
Hi @dymczykm ;
Thank you very much for your quick response; following your advice I commented the timestamp requirement in the file imu-measurements-buffer-inl.h like this:
```
  // Enforce strict time-wise ordering.
/*
  vio::ImuMeasurement last_value;
  if (buffer_.getNewestValue(&last_value)) {
    CHECK_GT(timestamp_nanoseconds, last_value.timestamp)
        << "Timestamps not strictly increasing.";
  }
*/
```
I then compiled again and then the error changes: 

[image]

Could it be something strange in recording/publishing ros messages? Im using the recommended way Tango ROS Streamer app. Ros version is kinetic.

Thank you very much in advance

MetaI, 2018-03-05:
Hi again @dymczykm
I checked Android output, modified Tango Ros Streamer to log header timestamp for imu
```
1520260756:99000000
03-05 15:39:16.106 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:105000000
03-05 15:39:16.110 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:110000000
03-05 15:39:16.114 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:114000000
03-05 15:39:16.119 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:119000000
03-05 15:39:16.125 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:124000000
03-05 15:39:16.131 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:131000000
03-05 15:39:16.135 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:135000000
03-05 15:39:16.140 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:140000000
03-05 15:39:16.145 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:145000000
03-05 15:39:16.150 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:149000000
03-05 15:39:16.157 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:157000000
03-05 15:39:16.165 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:165000000
03-05 15:39:16.167 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:166000000
03-05 15:39:16.169 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:169000000
03-05 15:39:16.176 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:175000000
03-05 15:39:16.179 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:179000000
03-05 15:39:16.184 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:184000000
03-05 15:39:16.189 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:189000000
03-05 15:39:16.199 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:199000000
03-05 15:39:16.205 24825-24825/eu.intermodalics.tango_ros_streamer I/ImuNode: 1520260756:205000000
```

I will try to check if the format is correct in the suscribers side. Any ideas are welcome!

MetaI, 2018-03-08:
All right, it seems that Header format in kinetik and lunar is broken for Tango ROS Streamer, or at least values for maplab aren't converted so well to Eigen (also had issuest with mesh reconstruction, but that is another issue). I found that using Indigo and Eigen versions that came with Ubuntu 14.04 everything works!. I will try to investigate further the specific reason, but it seems this specific software stack is good to go

dymczykm, 2018-03-12:
@MetaI so after switching to Ubuntu 14.04 everything works as expected?

MetaI, 2018-03-15:
@dymczykm I think the combination of Rosjava library in ROS publisher side(Tango ROS Streamer Android app) with ROS Indigo works. Indigo version binary packages are available in Ubuntu 14.04  and not in Ubuntu16.04. I still want to try compiling Indigo in newer versions of Ubuntu to confirm this conjectures.
I also don't know if Eigen library is involved or not. I had a lot of issues with kinetik that are easily replicable with any tango device + tango ros streamer and kinetik or lunar.  I'm using the Lenovo Phab 2 Pro device and the Fisheye camera right now

Thanks again @dymczykm

jian-li, 2018-05-08:
Does the tango works well on Maplab? Thx!

jian-li, 2018-05-19:
I tested the tango with ROS, the data was transfered by wifi, the delay is changing. And failed to run the rango transfered data on vins mono, if anyone make it, please tell me. thank you!

dymczykm, 2018-05-21:
@jian-li Could you share a bit more details of your system and how do you transfer the data? What is exactly 'vins-mono' (I assume that's not a part of maplab)?

jian-li, 2018-05-22:
Em, there is a ros package for streaming Tango data-tango_ros_streamer. Vins-mono is an open-source monocular visual inertial system, https://github.com/HKUST-Aerial-Robotics/VINS-Mono.

dymczykm, 2018-07-02:
@jian-li I'm afraid we can't provide any support for vins-mono.

MetaI, 2019-01-03:
Hi @dymczykm, @jian-li,
After struggling a lot with this device, I think I finally got it working well with I few conclussions:
As discovered here #63, lowering the imu rate to 50Hz in the streamer part, and disabling health checking in rovioli (depending on the case, but most of the time the best way was to disable it) made the estimation work quite well. It's not a perfect solution, but it will allow you to use this device. Also for the delays I thing you could use usb thetering or perhaps implement a rosjava node locally depending on your needs. I didn't test vins-mono sorry

closed 2019-01-03 by MetaI

gakkiHp, 2019-03-08:
@MetaI 
Hi, I have the same problem as you.How do you recompile after you annotate the following function? Catkin build vi - common? This doesn't seem to work.
  // Enforce strict time-wise ordering.
/*
  vio::ImuMeasurement last_value;
  if (buffer_.getNewestValue(&last_value)) {
    CHECK_GT(timestamp_nanoseconds, last_value.timestamp)
        << "Timestamps not strictly increasing.";
  }
*/