ethz-asl/maplab #49: Map recorded by ROVIOLI seems very sparse

iliis, 2018-03-05:
I'm trying to do dense reconstruction of a room using maplab/ROVIOLI with a Intel Realsense ZR300. 
ROVIOLI is running fine with the ZR300, but the map that is generated in the end only contains very few keyframes. Maplab fails to detect any loop closures and I don't see how this map could be used to reconstruct a dense map given that there are so few frames either.

This is the recoding I made (by walking up and down the corridor in our lab and doing a few circles in some of the offices)
```
$ rosbag info hallway3.bag
path:        hallway3.bag
version:     2.0
duration:    10:04s (604s)
start:       Feb 28 2018 17:03:22.89 (1519833802.89)
end:         Feb 28 2018 17:13:27.11 (1519834407.11)
size:        20.7 GB
messages:    341535
compression: none [17991/17991 chunks]
types:       cuckoo_time_translator/DeviceTimestamp [4e34c15f54cbfa3f64408fe764e59ff4]
             dynamic_reconfigure/Config             [958f16a05573709014982821e6822580]
             dynamic_reconfigure/ConfigDescription  [757ce9d44ba8ddd801bb30bc456f946f]
             sensor_msgs/CameraInfo                 [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/Image                      [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu                        [6a62c6daae103f4ff57a132d6f95cec2]
topics:      /zr300_node/color/camera_info                     17971 msgs    : sensor_msgs/CameraInfo                
             /zr300_node/color/image_raw                       17970 msgs    : sensor_msgs/Image                     
             /zr300_node/device_time                          120233 msgs    : cuckoo_time_translator/DeviceTimestamp
             /zr300_node/device_time/parameter_descriptions        1 msg     : dynamic_reconfigure/ConfigDescription 
             /zr300_node/device_time/parameter_updates             1 msg     : dynamic_reconfigure/Config            
             /zr300_node/fisheye/camera_info                   18025 msgs    : sensor_msgs/CameraInfo                
             /zr300_node/fisheye/image_raw                     18025 msgs    : sensor_msgs/Image                     
             /zr300_node/imu                                  149309 msgs    : sensor_msgs/Imu
```

Maplab tells me that there are only 140 nodes in the map:
```
maplab <hallway3>:/$ ms
Mission statistics: 

Mission 0:          		c3b971e3ffffffff0f00000000000000	Viwls
	NCamera Sensor:     	c6a6f6c779e1b09cf49f11629787945c
	IMU Sensor:         	c63aecb41babcabc7e1fac37c7cbe7bf
	Vertices:           	140                      
	Landmarks:          	7807                     
	Landmarks by first observer backlink:	                         
	Camera 0:           	4740 (g:66 b:4674 u:0)   
	Camera 1:           	3067 (g:22 b:3045 u:0)   
	Observations:       	32427                    
	Num edges by type:  	                         
	IMU:                	139                      
	Wheel-odometry:     	0                        
	Loop-closure:       	0                        
	Distance travelled [m]:	433.511871               
	Start to end time:  	2018_03_02_16_28_50 to 2018_03_02_16_28_50
	T_G_M               	unknown  
```

Also, there are only 280 frames in that map:
```  
$ ls resources/raw_images | wc -l
280
```  

This is the resulting track:
[image]

The command I used to run the odometry:
```
#!/usr/bin/env bash

my_config_dir="/home/samuel/calibrations/zr300/calib4_imu/maplab"

LOCALIZATION_MAP_OUTPUT="/home/samuel/maps/"
ROSBAG=$1
NCAMERA_CALIBRATION="$my_config_dir/ncamera.yaml"
IMU_PARAMETERS_MAPLAB="$my_config_dir/maplab_imu.yaml"
IMU_PARAMETERS_ROVIO="$my_config_dir/rovio_imu.yaml"
REST=$@

rosrun rovioli rovioli \
  --alsologtostderr=1 \
  --v=2 \
  --ncamera_calibration=$NCAMERA_CALIBRATION  \
  --imu_parameters_maplab=$IMU_PARAMETERS_MAPLAB \
  --imu_parameters_rovio=$IMU_PARAMETERS_ROVIO \
  --datasource_type="rosbag" \
  --save_map_folder="$LOCALIZATION_MAP_OUTPUT" \
  --optimize_map_to_localization_map=false \
  --map_builder_save_image_as_resources=true \
  --datasource_rosbag=$ROSBAG $REST
```

What am I doing wrong here? Do I have to change the parameters for ROVIO (in rovio_default_config.info)? Or am I supposed to record the `/maplab_rovio/T_G_I` topic myself?

mfehr, 2018-03-05:
Hi @iliis 
Looks like something went wrong in ROVIOLI while building the map, but I can't see anything obviously wrong with what you're doing based on the information I have. Most likely something is wrong with the timing. The number of images on the other hand seems to nicely correspond to the vertices, so that's fine. Could you provide us with some additional data:
 * Could you upload the the ROVIOLI log output somewhere for us to see?
 * Do you use our [maplab zr300 ros node](https://github.com/ethz-asl/maplab_realsense)?
 * How did you calibrate the sensor?

iliis, 2018-03-05:
Thanks for the quick reply!

* Sure, I put the log here: https://gist.githubusercontent.com/iliis/4843665616b6fb533d9710fa018504fe/raw/70b195ced9381cb08ab90c6217e590872d159e61/gistfile1.txt
There are a lot of messages like
`I0305 16:02:27.517788 20486 OutlierDetection.hpp:47] Detected outlier - mahalanobis (d / threshold):29.0232 > 9.21 - count : 1.`
* Yes, I've used the maplab_realsense node. The example I gave is running from a bagfile recording of the maplab_realsense node, altough running it live results in the same output.
* I calibrated the ZR300 using Kalibr and an apriltag. I've used the IMU intrinsics from http://robotics.ethz.ch/%7Easl-datasets/maplab/calibrations/vi_sensor_intel_realsense_zr300/ together with Kalibr's IMU-camera calibration for the extrinsics.

Again, the odometry itself seems to works fine. I get a nice and continuous pose on `/maplab_rovio/T_G_I`

dymczykm, 2018-03-05:
Hi @iliis,
looking at your map statistics, most of the landmarks were classified as bad (only 66 out of 4740 are good, usually that's at least 20%):
``Camera 0:           	4740 (g:66 b:4674 u:0) ``
that's why you only see so few of them. That would suggest many of the landmarks do not fulfill the criteria that are related to min/max distance from the observer, min number of observers per landmark or the disparity angle of the bearing vectors.

Looking at the trajectory, are you sure this is a correct path? Looks extremely noisy to me (and might cause the issue with landmarks).

iliis, 2018-03-06:
The path looks correct, yes. I was walking up and down the hallway and did a few circles in two of the offices (one on the left and one middle right). The appearant noisyness comes from the fact that there are very few vertices created (only one every few meters). It looks to me like the odometry itself is working well. You can see some drift tough: The long straight lines should overlap.

iliis, 2018-03-06:
Here's a visualization with a manually aligned floor plan:
[image]

dymczykm, 2018-03-07:
@iliis Honestly, that doesn't look good to us. We noticed that you're located somewhere nearby, would you have time to simply step by and look at your issue together?

iliis, 2018-03-07:
Hehe, that would be very helpful, yes! I've set it up on my PC in the lab tough, so let me try to get everything running on my laptop. Otherwise, I'll just bring the raw bagfile.
So, i'll just visit you at ASL, ok?

iliis, 2018-03-09:
For anyone stumbling upon this, we found a workaround: Make sure there is only a single camera defined in your `ncamera.yaml`. (I had both the fisheye and the color camera from the ZR300 in there.)

mfehr, 2018-03-09:
@iliis In your case of the ZR300 that's certainly correct, in general having more than one fisheye camera should not be a problem , we do this for VI-sensor datasets all the time. ROVIO will still use only one, but ROVIOLI will run feature tracking and map-building based on both/all of the cameras. I think the issue must have been the fact that a color image was passed into our feature tracker, this must have lead to some weird side effects. We haven't investigated whether the problem was the significantly larger size of the color image or if the color image was corrupted and/or misinterpreted as gray scale. It might have worked if we convert the image to gray scale.

closed 2018-03-09 by mfehr

iliis, 2018-03-09:
Hm, I don't think it was the image size, as both the fisheye and the color camera of the ZR300 only provide 640 x 480 pixels. I could try converting it to gray scale tough. Could it be a timing issue after all? The cameras on the VI-sensor are accurately synchronized while the ones on the ZR300 might not be.

mfehr, 2018-03-10:
Ah I thought the RGB image was bigger for the ZR300. That's a very good point. I think you are right about the synchronization, it would explain the sparsity of the vertices. ROVIOLI probably tries to synchronize VisualNFrames (a struct that contains the camera images of all cameras and data for one specific point in time), but rarely succeeds, i.e. only if the RGB timestamp happens to be within a certain range of the Fisheye.

NikolausDemmel, 2018-03-10:
> Hm, I don't think it was the image size, as both the fisheye and the color camera of the ZR300 only provide 640 x 480 pixels. I could try converting it to gray scale tough. Could it be a timing [...]

AFAIK the fisheye tracking camera is time synchronized with the IMU, while the RGB camera is not. Also, I believe the color camera is rolling shutter, wheres the tracking camera is global shutter.