ethz-asl/maplab #51: How do I stop and save the recorded data when I Running ROVIOLI in VIO mode ???

linhuincu, 2018-03-12:
Question: How do I stop and save the recorded data when I Running ROVIOLI in VIO mode ???

My question is simple:

My maplab is run well, and My question is  when I Running ROVIOLI in VIO mode follow https://github.com/ethz-asl/maplab/wiki/Running-ROVIOLI-in-VIO-mode, 

when I use rosbag model, after run the MH_01.bag, it have save the map in the aim path, but when I use the rostopic model to run the rosbag and run the realsense ZR300 , after the terminal0 was stop, the terminal2 always shows " No IMU messages  ", " No camera messages ". the two terminal 2 can't stop and can't save the map in the aim path. 

SO, My question is : 
1. When I Running ROVIOLI in VIO mode , How do I save the record data of terminal 2  ?
2. When I Running ROVIOLI in VIO mode , How do I stop terminal 2 instead of ctrl + C ?

What should I do? which did I wrong?
very thank you!

my command is:

<1> ###from a rostopic:

#terminal0:
roscore&
#termina1：
rosbag play /home/turtle/DataSet/EuRoC_datasets/Rosbag/MH_01_easy.bag
#terminal2:
source ~/maplab_ws/devel/setup.bash
rosrun rovioli tutorial_euroc_live /home/turtle/maplab_ws/output/

<2> ###from realsense ZR300:

#terminal1：
source ~/maplab_ws/devel/setup.bash
sudo modprobe uvcvideo
roslaunch maplab_realsense maplab_realsense.launch
#terminal2:
source ~/maplab_ws/devel/setup.bash
rosrun rovioli tutorial_realsense_live /home/turtle/maplab_ws/output

---

Before submitting this issue, please consider the following **checklist**:

 YES - Consult the [FAQ](https://github.com/ethz-asl/maplab/wiki/FAQ) and [Known Issues](https://github.com/ethz-asl/maplab/wiki/Known-Issues) pages.
 YES - Make sure maplab was installed using the proper [installation instructions](https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu)
 - For bugs and build problems please add (if applicable):
   - Version of:
     - Operating System (32/64bit, 14.04/16.04/...): 64bit, 16.04
     - Eigen: 3.2
     - ROS: kinetic
   - A backtrace (if maplab crashed): No
   - Run `catkin config` in your workspace and post it here.

root@turtle:~# catkin config
-------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /home/turtle
-------------------------------------------------
Source Space:      [missing] /home/turtle/src
Log Space:         [missing] /home/turtle/logs
Build Space:       [missing] /home/turtle/build
Devel Space:       [missing] /home/turtle/devel
Install Space:      [unused] /home/turtle/install
DESTDIR:            [unused] None
-------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------

-------------------------------------------------
WARNING: Workspace `/home/turtle` is not yet
initialized. Use the `catkin init` or run
`catkin config --init`.
-------------------------------------------------
WARNING: Source space `/home/turtle/src` does
not yet exist.
-------------------------------------------------

   - Details about the commands/scripts used.
   - Details about special parameter values or manual changes to the system.
i5-3210, GT635, 4 core, lenovo M490
   - Pictures/screen recordings
 - For general questions:
   - Please elaborate, we are no mind readers :)

mfehr, 2018-03-12:
@linhuincu the proper way to abort ROVIOLI in live mode is `Ctrl+C`, we are catching this signal and storing the map. If this doesn't work we would need some console output of ROVIOLI to help you. Could you post it here, or upload it somewhere for us to see?

mfehr, 2018-03-29:
Related issue https://github.com/ethz-asl/maplab/issues/58 (?)