ethz-asl/maplab #52: when I run the realsense ZR300 rosnode, it shows the timestamp have something wrong......

linhuincu, 2018-03-12:
I'm follow the https://github.com/IntelRealSense/librealsense/blob/v1.12.0/doc/installation.md and https://github.com/ethz-asl/maplab_realsense/blob/master/README.md to install the realsense ZR300, and use it to Running ROVIOLI in VIO mode, it works well besides the timestamp.
Does Some one could give me some solve advice ?
thank you!

my terminal command is:
########################
###from realsense zr300:

#terminal1:
source ~/maplab_ws/devel/setup.bash
rosrun rovioli tutorial_realsense_live /home/turtle/maplab_ws/output

#terminal2：
source ~/maplab_ws/devel/setup.bash
sudo modprobe uvcvideo
roslaunch maplab_realsense maplab_realsense.launch
#########################

the error is:
#######################

E0312 21:25:33.540139 24448 zr300.cc:490] The timestamp of frame 3 of type INFRARED did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
E0312 21:25:33.572222 24448 zr300.cc:490] The timestamp of frame 4 of type INFRARED did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
E0312 21:25:33.593979 24448 zr300.cc:490] The timestamp of frame 5 of type COLOR did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
E0312 21:25:33.626953 24448 zr300.cc:490] The timestamp of frame 7 of type COLOR did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
E0312 21:25:33.660995 24448 zr300.cc:490] The timestamp of frame 9 of type COLOR did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
E0312 21:25:33.692926 24448 zr300.cc:490] The timestamp of frame 11 of type COLOR did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
E0312 21:25:33.733770 24448 zr300.cc:490] The timestamp of frame 9 of type INFRARED did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
E0312 21:25:33.733849 24448 zr300.cc:490] The timestamp of frame 9 of type INFRARED2 did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
E0312 21:25:33.760748 24448 zr300.cc:490] The timestamp of frame 13 of type COLOR did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
................
................
#########################

---

Before submitting this issue, please consider the following **checklist**:

 YES - Consult the [FAQ](https://github.com/ethz-asl/maplab/wiki/FAQ) and [Known Issues](https://github.com/ethz-asl/maplab/wiki/Known-Issues) pages.
 YES - Make sure maplab was installed using the proper [installation instructions](https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu)
 - For bugs and build problems please add (if applicable):
   - Version of:
     - Operating System (32/64bit, 14.04/16.04/...): 64bit, 16.04
     - Eigen: 3.2
     - ROS: kinetic
   - A backtrace (if maplab crashed): No
   - Run `catkin config` in your workspace and post it here.

root@turtle:~# catkin config
-------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /home/turtle
-------------------------------------------------
Source Space:      [missing] /home/turtle/src
Log Space:         [missing] /home/turtle/logs
Build Space:       [missing] /home/turtle/build
Devel Space:       [missing] /home/turtle/devel
Install Space:      [unused] /home/turtle/install
DESTDIR:            [unused] None
-------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------

-------------------------------------------------
WARNING: Workspace `/home/turtle` is not yet
initialized. Use the `catkin init` or run
`catkin config --init`.
-------------------------------------------------
WARNING: Source space `/home/turtle/src` does
not yet exist.
-------------------------------------------------

   - Details about the commands/scripts used.
   - Details about special parameter values or manual changes to the system.
i5-3210, GT635, 4 core, lenovo M490
   - Pictures/screen recordings
 - For general questions:
   - Please elaborate, we are no mind readers :)

mfehr, 2018-03-12:
@linhuincu So the error message you are seeing means that the ROS node that talks to the sensor is unable to get infrared and RGB images that have been timestamped with the microcontroller clock. [This](https://github.com/ethz-asl/maplab_realsense/blob/master/src/zr300.cc#L51) call should tell your ZR300 that you would like to have all data properly timestamped, but it looks like that's not happening on your device. Do you happen to have an early version of the ZR300? 

I'm not sure how to solve this, if it can be solved at all without tinkering with the sensor software. However, if you are only interested in running state estimation on the ZR300 you can simply turn of IR, RGB and depth publishing in the launch file ([here and below](https://github.com/ethz-asl/maplab_realsense/blob/master/launch/maplab_realsense.launch#L17)). Alternatively you can make a small change to the ROS node by commenting out [this](https://github.com/ethz-asl/maplab_realsense/blob/master/src/zr300.cc#L488) check such that it publishes the IR/RGB/Depth data even if they are not properly timestamped.

bsbretly, 2018-07-08:
@mfehr - I'm having a similar issue with my ZR300 as well, though with the fisheye timestamp. I imagine this will result in poor VIO results using ROVIOLI.

When launching maplab_realsense.launch I get the following error:
E0708 16:50:47.415486 19441 zr300.cc:568] The timestamp of frame 215 of type FISHEYE did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!

Any suggestions on troubleshooting this issue? 

Thanks!

dymczykm, 2018-07-09:
@bsbretly does this happen for all frame types and all frames or only once in a while?

bsbretly, 2018-07-09:
@dymczykm this only happens once in a while.

dymczykm, 2018-07-09:
@bsbretly that shouldn't matter that much then. The wrongly timestamped frames are rejected, but all the others are OK and consumed by the estimator. I don't think this would cause much difference in the VIO output.

bsbretly, 2018-07-09:
Thanks for the help @dymczykm. I tested this earlier today and it seems to estimate pose quite well despite the error. 

Appreciate your input!

closed 2018-08-15 by mfehr

maxwu8416, 2019-06-21:
hi @mfehr
yesterday i have the similar error and i already check your suggest link to see the code but not really know what your mean i think maybe the line you marked is already change from the new update.
i was opened a new issue of my error proplem at here
[https://github.com/ethz-asl/maplab/issues/152](url)