ethz-asl/maplab #53: Is there a way to publish real-time point cloud?

Doodleshr, 2018-03-13:
Hi,

I am trying to use only the front end of the framework and build a voxel map from generated point cloud in real time, but from my understanding of the framework it seems that publishing point cloud is not currently supported from provided WIKIs. If it is the case, is there a way to realize this function?

Thanks for your help
shr

mfehr, 2018-03-13:
Hi @Doodleshr 

You're right, the framework is lacking real-time dense reconstruction capabilities. However, we have used ROVIOLI successfully for this purpose before. There are two options:
 * You have an RGB-D camera and get pointclouds:
    *  Just combine the ROVIOLI node with a voxblox node.
 * You have a stereo camera:
     * Combine the ROVIOLI node with a stereo matcher node and a voxblox node.

Have a look at these links:
https://github.com/ethz-asl/voxblox/blob/master/README.md#running-voxblox
https://github.com/ethz-asl/image_undistort#dense_stereo_node

Hope this helps

Doodleshr, 2018-03-14:
@mfehr thanks for your help!

closed 2018-03-14 by Doodleshr

plumewind, 2018-07-25:
Hi @Doodleshr 

I am also doing related work now. I plan to use a depth camera to get the point cloud and write a ROS node to complete the construction of the voxel map.

However, the specific process details are somewhat unclear. My general idea is like that :+1: 
(1) get the point cloud from the depth camera
(2) send it to the ROVIOLI node to get the VI map
(3) finally send it to voxblox to get the TSDF grid map (voxel map). 
Is this process correct?  @mfehr and anyone , thank you very much.