ethz-asl/maplab #54: VIO suddenly starts to accelerate then diverges eventually

BryceeeeeChen, 2018-03-14:
Hi,
When I ran ROVIOLI in VIO mode to generate VI map, sometimes the VIO would suddenly accelerate after good tracking for a while (like this, https://www.dropbox.com/s/62pvf3rojure73y/suddenly%20accelerates%20then%20diverges.MOV?dl=0). Was that caused by the noisy accelerometer data? I also provide figures generated by plot_vi_states_of_mission. 
Before batch optimization:
[image]
After batch optimization:
[image]
I would be thankful for any help!

NikolausDemmel, 2018-03-14:
I am running the original ROVIO from https://github.com/ethz-asl/rovio and I have sometimes noticed the same thing. I have yet to investigate more closely, but I suspect that it might have to do with the parameterization of IMU noise.

It doesn't happen for all my test sequences of various length and kinds of motion (though all of them are with a handheld camera rig, so I suspect that the movement is never too degenerate as all degrees of freedom seem to be excited at least a little bit when walking naturally). I haven't really been able to find a correlation with the kind of motion where this happens.

My explanation would be that ROVIO gets too certain about the orientation estimate that is actually wrong. Once you have a wrong orientation, the gravity is subtracted wrongly from accelerometer measurements and a rapidly accelerating estimator state is the consequence.

I guess it could be either the orientation itself where the estimator is overconfident, or the gyro or accelerometer biases, so I'm not sure without further investigation, which of the IMU noise parameters needs to be tuned, probably all of them.

Would be interesting to hear if others agree with this assessment.

dymczykm, 2018-03-14:
@BryceeeeeChen your video suggests that your sensor is mounted on a wheeled robot? is that true?

BryceeeeeChen, 2018-03-14:
Hi, @NikolausDemmel. IMHO, the wrong subtraction of gravity that you pointed out would definitely drift the system. But I guess it should only induce constantly slow drift to one direction instead of suddenly aggressive acceleration and divergence. Maybe slow drift like this https://github.com/ethz-asl/rovio/issues/140, on which I used my own calibration parameters obtained by kalibr and ASD charts (not good enough, which might cause bad orientation estimation). Actually, I have rarely encountered this aggressive acceleration problem since I started testing ROVIOLI with handheld ZR300 using the calibration parameters provided by ASL. 

It was likely to happen when 1) Lighting condition significantly change. 2) mount it on a wheeled robot. SO, @dymczykm you are right, I'm mounting the camera on a heavy wheeled robot. Is that an issue? BTW, the robot moves quite smoothly and steadily, it is not like a turtlebot always wavering in motion.

NikolausDemmel, 2018-03-14:
I guess "smooth and steady" motion might be an issue, since you don't have any angular velocities and no acceleration when you just translate with constant speed.

NikolausDemmel, 2018-03-14:
Oh, or it might also be due to inaccurate extrinsic calibration between camera and IMU. Since that is kept constant, a systematic drift seems like a reasonable consequence.

BryceeeeeChen, 2018-03-22:
The cause of this issue is illustrated by @NikolausDemmel 's reply with the thump up logo and @mfehr 's reply in #56. Need to integrate wheel odometry to solve this scale issue. Close now.

closed 2018-03-22 by BryceeeeeChen

YonatanSimson, 2021-07-04:
I notice this happens wen I try initialize the drone in the air. In this case I the inertial frame ans W the internal frame of ROVIO are badly mis aligned. PErhaps the solution would be to add gravity to the state like in msf_ekf? @NikolausDemmel