ethz-asl/maplab #58: cann't save the map in VIO mode

wendaoyuchen, 2018-03-28:
when I want to build a map from MH_01_easy.bag in VIO mode , it can run well but there is nothing in the save_folder. I start two terminals, one for roscore, one for runing the dataset,
rosrun rovioli tutorial_euroc /home/liuzhen/test/maplab_test/ `/home/liuzhen/test/maplab_test/MH_01_easy.bag`
when the bag is over , it shows some messages in the terminal as follow,
> W0328 10:24:36.171046  9697 imu-camera-synchronizer.cc:103] No IMU messages have been received in the last 5 seconds. Check for measurement drops or if the topic is properly set in the maplab IMU configuration file
and ctrl+C cann't stop it, so I have to close the window which shows the picture in the MH_01_easy.bag, but I open the save_folder, there is just a folder named _0, and the sub_folder is named resources, but there is nothing in them.

mfehr, 2018-03-28:
Hi @wendaoyuchen 
Sounds like the end-of-rosbag signal doesn't shutdown ROVIO properly and therefore fails to trigger the saving of the map. Could you provide some additional information?
 * increase the verbosity level (--v=3) and check if you missed some event happening at the end of the rosbag.
 * run ROVIOLI in live mode, i.e. play the rosbag + run ROVIOLI
 * please provide us with some relevant information about your system (see the issue checklist)

wendaoyuchen, 2018-03-28:
Thanks for your reply, @mfehr 
1.you mean that increasing the verbosity level(--v=3) is like that:
`roscore& rosrun rovioli tutorial_euroc /home/liuzhen/test/maplab_test/ /home/liuzhen/test/maplab_test/MH_01_easy.bag --v=3`
2. I used the "Building a map from a rostopic"mode, too. Is that ROVIOLI live mode? But it didn't work.
3.I try ctrl+\, it can stop it, but there is nothing in the folder.the output is as follow,
```
^\/home/liuzhen/maplab_ws/src/maplab/applications/rovioli/scripts/tutorials/tutorial_euroc: line 22:  9538 Quit                    (core dumped) rosrun rovioli rovioli --alsologtostderr=1 --v=2 --ncamera_calibration=$NCAMERA_CALIBRATION --imu_parameters_maplab=$IMU_PARAMETERS_MAPLAB --imu_parameters_rovio=$IMU_PARAMETERS_ROVIO --datasource_type="rosbag" --save_map_folder="$LOCALIZATION_MAP_OUTPUT" --optimize_map_to_localization_map=false --map_builder_save_image_as_resources=false --datasource_rosbag=$ROSBAG $REST
[2]   Exit 1                  roscore
[3]+  Exit 1                  roscore
```
Thanks or your help, I guess the ctrl+c cann't be captured by the project, but I don't know why.

mfehr, 2018-03-28:
1. This should work, but if you want to be sure, just add it to the flags inside the script. Could you start roscore in a separate window? Maybe Ctrl-C is intercepted by the roscore and not ROVIOLI.
2. That's the one, yes. What is the behavior there? In live mode ROVIOLI will save the map as soon as you abort the estimator with Ctr-C.
3. We do capture ctrl-c using the ROS API, but maybe there is sth about your ros distro/setup that doesn't do that. What ROS version do you have?

Could you upload your console output (with increased verbosity) somewhere and share it with us?

wendaoyuchen, 2018-03-29:
Thanks for your reply! My ROS version is Kinetic,
1.I tried some methods to run it, strt roscore in another terminal, change the save_folder, change the dataset,change the verbosity level, but all of them didn't work.
2.I open three terminals to run it, one for roscore, the input is just roscore, then I input
`rosrun rovioli tutorial_euroc_live /home/liuzhen/test/maplab_test/ --v=3`
in the second terminal, the output in the window is shown as follow,
```
liuzhen@liuzhen:~$ rosrun rovioli tutorial_euroc_live /home/liuzhen/test/maplab_test/ --v=3
I0329 10:16:00.235033 16398 rovio-factory.cc:169] Loading ROVIO configuration template: /home/liuzhen/maplab_ws/src/maplab/applications/rovioli/share//rovio_default_config.info
Set distortion parameters (Radtan) to: k1(-0.283408), k2(0.0739591), k3(0), p1(0.000193595), p2(1.76187e-05)
Registered filter state update callback.
I0329 10:16:00.419657 16398 map-builder-flow.cc:31] Set VIMap folder to: /home/liuzhen/test/maplab_test/_4
I0329 10:16:00.451215 16398 rovioli-node.cc:105] Starting data source...
I0329 10:16:00.451288 16398 rovioli-app.cc:136] 
Message delivery queues:
subscriber-node               queue-topic                   queue-id                      num elements                  
MapBuilderFlow                LOCALIZATION_RESULT           142d..0000                    0                             
MapBuilderFlow                ROVIO_ESTIMATES               8928..0000                    0                             
DataPublisherFlow             ROVIO_ESTIMATES               ed3d..0000                    0                             
ImuCameraSynchronizerFlow     IMAGE_MEASUREMENTS            13f3..0000                    0                             
MapBuilderFlow                TRACKED_NFRAMES_AND_IMU       4023..0000                    0                             
DataPublisherFlow             VIO_UPDATES                   994a..0000                    0                             
RovioFlow                     LOCALIZATION_RESULT           2375..0000                    0                             
RovioFlow                     IMU_MEASUREMENTS              623e..0000                    0                             
DataPublisherFlow             LOCALIZATION_RESULT           2d39..0000                    0                             
MapBuilderFlow                VIO_UPDATES                   f316..0000                    0                             
ImuCameraSynchronizerFlow     IMU_MEASUREMENTS              1cf8..0000                    0                             
FeatureTrackingFlow           SYNCED_NFRAMES_AND_IMU        2e5d..0000                    0                             
DataPublisherFlow             RAW_VIMAP                     9c28..0000                    0                             
RovioFlow                     IMAGE_MEASUREMENTS            4966..0000                    0                             
FeatureTrackingFlow           ROVIO_ESTIMATES               916e..0000                    0                             
SyncedNFrameThrottlerFlow     TRACKED_NFRAMES_AND_IMU       a891..0000                    0                             

```
And the input in the last terminal is 
`rosbag play test/maplab_test/MH_02_easy.bag `
and when it run over, the third wondow will stop automatically.
the second window will output 

```
^CI0329 10:12:44.652045 11421 rovioli-node.cc:110] Closing data source...
I0329 10:12:44.652140 11421 map-builder-flow.cc:111] Initializing landmarks of created map.
I0329 10:12:45.142377 11421 landmark-triangulation.cc:248] Getting vertices of mission: dda7..0000
I0329 10:12:45.145512 11421 landmark-triangulation.cc:253] Retriangulating landmarks of 1516 vertices.
```
and then it will repeat to output
`W0329 10:21:15.383613 16424 imu-camera-synchronizer.cc:103] No IMU messages have been received in the last 5 seconds. Check for measurement drops or if the topic is properly set in the maplab IMU configuration file`
here ctrl+C doesn't work, I have to input ctrl+\ to interrupt it, and the output is 
`^\/home/liuzhen/maplab_ws/src/maplab/applications/rovioli/scripts/tutorials/tutorial_euroc_live: line 20: 16398 Quit                    (core dumped) rosrun rovioli rovioli --alsologtostderr=1 --v=2 --ncamera_calibration=$NCAMERA_CALIBRATION --imu_parameters_maplab=$IMU_PARAMETERS_MAPLAB --imu_parameters_rovio=$IMU_PARAMETERS_ROVIO --datasource_type="rostopic" --save_map_folder="$LOCALIZATION_MAP_OUTPUT" --map_builder_save_image_as_resources=false --optimize_map_to_localization_map=false $REST`
so I am confused about it, Is that neccessary to modify the code?Or I should try maplab in another computer?

mfehr, 2018-03-29:
Can you try running rovioli with `--num_hardware_threads=1` to get rid of the multithreading?
If 
```
I0329 10:12:45.145512 11421 landmark-triangulation.cc:253] Retriangulating landmarks of 1516 vertices.
```
is really the last message you get before it keeps repeating the "No IMU" message, then it got stuck in retriangulating the landmarks or saving the map. Both are using this `common::ParallelProcess` function for multithreading. It is most likely a bug.
 * If the flag fixes your problem then it's likely a multithreading bug that seems to thrive in your setup. To the best of my knowledge we haven't seen this one in any of our machines here. This might make it very difficult to help you.
 * Other things you can try yourself first:
   * simply write some log output [here](https://github.com/ethz-asl/maplab/blob/29b906cd55716beb2c33e32d43ca84a7abcc99ce/applications/rovioli/src/map-builder-flow.cc#L125) to be able to tell if it's retriangulation or map saving that acts up
   * compile in Debug or RelWithDebugInfo mode and use gdb (and provide us with the backtrace). I copied and adapted a wiki page from our internal wiki here with some hints on how to run the debugger: [Compilation and Debugging](https://github.com/ethz-asl/maplab/wiki/Compilation-and-Debugging)
   * different machine or VM with a (fresh) Ubuntu LTS.

wendaoyuchen, 2018-07-19:
Hi, recently I try to figure out this problem. I set serval output to find where is the problem.
And this common::ParallelProcess in parallel-process.h is stuck in this loop
`CHECK_EQ(threads.size(), blocks.size());
  for (size_t block_idx = 0; block_idx < blocks.size(); ++block_idx) {
    threads[block_idx].join();
  }`
I set some output like this
`  std::cerr <<"ParallelProcess: here is OK3"<<std::endl;
  CHECK_EQ(threads.size(), blocks.size());
  for (size_t block_idx = 0; block_idx < blocks.size(); ++block_idx) {
    threads[block_idx].join();
  }
      std::cerr <<"ParallelProcess: here is OK4"<<std::endl;`
and the terminal output is shown as
`ParallelProcess: here is OK3
W0719 10:37:55.983767  3854 imu-camera-synchronizer.cc:103] No IMU messages have been received in the last 5 seconds. Check for measurement drops or if the topic is properly set in the maplab IMU configuration file
W0719 10:37:55.983892  3854 imu-camera-synchronizer.cc:111] No camera messages have been received in the last 5 seconds. Check for measurement drops or if the topic is properly set in the camera configuration file
`
Can you give some suggestion to solve it?
ps: I don't know how to use gdb to debug it,if gdb is neccessary, I will try it.

wendaoyuchen, 2018-07-19:
I add some additional output like below:
`std::cerr <<"ParallelProcess: here is OK3"<<std::endl;
  CHECK_EQ(threads.size(), blocks.size());
      std::cerr <<"ParallelProcess: thread size is "<<threads.size()<<std::endl;
      std::cerr <<"ParallelProcess: blocks size is "<<blocks.size()<<std::endl;
  for (size_t block_idx = 0; block_idx < blocks.size(); ++block_idx) {
    std::cerr <<"block_idx is "<<block_idx<<std::endl;
    threads[block_idx].join();
  }
      std::cerr <<"ParallelProcess: here is OK4"<<std::endl;`
and the output in terminal is
`ParallelProcess: here is OK3
ParallelProcess: thread size is 8
ParallelProcess: blocks size is 8
block_idx is 0
`

wendaoyuchen, 2018-07-19:
just now I make maplab on my laptop, and it can save the map,,,I'll try recompile maplab on this machine.