ethz-asl/maplab #59: Dose message_flow support "hot register" ?

pangfumin, 2018-03-28:
Hi, thank you for sharing this great SLAM frame work.
I want to use this framework to save vimap  in my vio  system.
However, different from what is done in ROVIOLI which register the feature-track and map-builder to message flow when initialize the whole system, I want to start featrue-track and map_builder when I push a buttton on my GUI after the vio has run for a while.
In this situation, I found  landmark_triangulation::(anonymous namespace)::retriangulateLandmarksOfVertex() is crashed.
> I0328 06:38:53.502740 32632 landmark-triangulation.cc:248] Getting vertices of mission: c364..0000
I0328 06:38:53.502995 32632 landmark-triangulation.cc:253] Retriangulating landmarks of 197 vertices.
I0328 06:38:54.002897 32632 landmark-triangulation.cc:248] Getting vertices of mission: c364..0000
I0328 06:38:54.003346 32632 landmark-triangulation.cc:253] Retriangulating landmarks of 197 vertices.
*** Error in `/persist/maplab_linux/devel/lib/ninebot_slam_demo/navigation-demo': double free or corruption (out): 0x00007fdc34017cd0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fdcc17827e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7fdcc178b37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7fdcc178f53c]
/persist/maplab_linux/devel/lib/libaslam_cv_common.so(_ZNSt6vectorIN10statistics18StatisticsMapValueESaIS1_EE19_M_emplace_back_auxIJS1_EEEvDpOT_+0x368)[0x7fdcbfee1f98]
/persist/maplab_linux/devel/lib/libaslam_cv_common.so(_ZN10statistics10Statistics9GetHandleERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x672)[0x7fdcbfee1682]
/persist/maplab_linux/devel/lib/libaslam_cv_common.so(_ZN10statistics18StatsCollectorImplC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0xc)[0x7fdcbfee170c]
/persist/maplab_linux/devel/lib/liblandmark_triangulation.so(+0xd2fe)[0x7fdcbc9a32fe]
/persist/maplab_linux/devel/lib/liblandmark_triangulation.so(+0xdd48)[0x7fdcbc9a3d48]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xb8c80)[0x7fdcc1da3c80]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7fdcc76736ba]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7fdcc18123dd]
======= Memory map: ========
00400000-0042c000 r-xp 00000000 08:01 10355761                           /persist/maplab_linux/devel/lib/ninebot_slam_demo/navigation-demo
0062b000-0062d000 r--p 0002b000 08:01 10355761                           /persist/maplab_linux/devel/lib/ninebot_slam_demo/navigation-demo
0062d000-0062e000 rw-p 0002d000 08:01 10355761                           /persist/maplab_linux/devel/lib/ninebot_slam_demo/navigation-demo
019ab000-02d0f000 rw-p 00000000 00:00 0                                  [heap]
7fdc24000000-7fdc24021000 rw-p 00000000 00:00 0 
7fdc24021000-7fdc28000000 ---p 00000000 00:00 0 
7fdc28000000-7fdc28071000 rw-p 00000000 00:00 0 
7fdc28071000-7fdc2c000000 ---p 00000000 00:00 0 
7fdc30000000-7fdc3007e000 rw-p 00000000 00:00 0 
7fdc3007e000-7fdc34000000 ---p 00000000 00:00 0 
7fdc34000000-7fdc34085000 rw-p 00000000 00:00 0 
7fdc34085000-7fdc38000000 ---p 00000000 00:00 0 
7fdc38000000-7fdc38247000 rw-p 00000000 00:00 0 
7fdc38247000-7fdc3c000000 ---p 00000000 00:00 0 
7fdc3c000000-7fdc3c962000 rw-p 00000000 00:00 0 
7fdc3c962000-7fdc40000000 ---p 00000000 00:00 0 
7fdc40000000-7fdc40021000 rw-p 00000000 00:00 0 
7fdc40021000-7fdc44000000 ---p 00000000 00:00 0 
7fdc44e7f000-7fdc48000000 rw-p 00000000 00:00 0 
7fdc48000000-7fdc48a86000 rw-p 00000000 00:00 0 
7fdc48a86000-7fdc4c000000 ---p 00000000 00:00 0 
7fdc4dffc000-7fdc4dffd000 ---p 00000000 00:00 0 
7fdc4dffd000-7fdc4e7fd000 rwxp 00000000 00:00 0 
7fdc4e7fd000-7fdc4e7fe000 ---p 00000000 00:00 0 
7fdc4e7fe000-7fdc4effe000 rwxp 00000000 00:00 0 
7fdc4effe000-7fdc4efff000 ---p 00000000 00:00 0 
7fdc4efff000-7fdc4f7ff000 rwxp 00000000 00:00 0 
7fdc4f7ff000-7fdc4f800000 ---p 00000000 00:00 0 
7fdc4f800000-7fdc50000000 rwxp 00000000 00:00 0 
7fdc50000000-7fdc509fd000 rw-p 00000000 00:00 0 
7fdc509fd000-7fdc54000000 ---p 00000000 00:00 0 
7fdc54000000-7fdc54021000 rw-p 00000000 00:00 0 
7fdc54021000-7fdc58000000 ---p 00000000 00:00 0 
7fdc58000000-7fdc58a03000 rw-p 00000000 00:00 0 
7fdc58a03000-7fdc5c000000 ---p 00000000 00:00 0 
7fdc5cb04000-7fdc5cb10000 r-xp 00000000 08:16 4332190                    /usr/lib/x86_64-linux-gnu/gtk-2.0/2.10.0/engines/libxfce.so
7fdc5cb10000-7fdc5cd10000 ---p 0000c000 08:16 4332190                    /usr/lib/x86_64-linux-gnu/gtk-2.0/2.10.0/engines/libxfce.so
7fdc5cd10000-7fdc5cd11000 r--p 0000c000 08:16 4332190                    /usr/lib/x86_64-linux-gnu/gtk-2.0/2.10.0/engines/libxfce.so
7fdc5cd11000-7fdc5cd12000 rw-p 0000d000 08:16 4332190                    /usr/lib/x86_64-linux-gnu/gtk-2.0/2.10.0/engines/libxfce.so
7fdc5d113000-7fdc5d114000 ---p 00000000 00:00 0 
7fdc5d114000-7fdc5d514000 rwxp 00000000 00:00 0 
7fdc5d514000-7fdc5d515000 ---p 00000000 00:00 0 
7fdc5d515000-7fdc5d915000 rwxp 00000000 00:00 0 
7fdc5d915000-7fdc5db15000 rw-p 00000000 00:00 0 
7fdc5db15000-7fdc5db2e000 r-xp 00000000 08:16 5141132                    /usr/lib/x86_64-linux-gnu/libtbbmalloc.so.2
7fdc5db2e000-7fdc5dd2d000 ---p 00019000 08:16 5141132                    /usr/lib/x86_64-linux-gnu/libtbbmalloc.so.2
7fdc5dd2d000-7fdc5dd2e000 r--p 00018000 08:16 5141132                    /usr/lib/x86_64-linux-gnu/libtbbmalloc.so.2
7fdc5dd2e000-7fdc5dd30000 rw-p 00019000 08:16 5141132                    /usr/lib/x86_64-linux-gnu/libtbbmalloc.so.2
[... 187 lines cut ...]
7fdca2cfb000-7fdca2efb000 ---p 00007000 08:16 4332394                    /usr/lib/x86_64-linux-gnu/libogg.so.0.8.2
7fdca2efb000-7fdca2efc000 r--p 00007000 08:16 4332394                    /usr/lib/x86_64-linux-gnu/libogg.so.0.8.2
7fdca2efc000-7fdca2efd000 rw-p 00008000 08:16 4332394                    /usr/lib/x86_64-linux-gnu/libogg.so.0.8.2
7fdca2efd000-7fdca2f07000 r-xp 00000000 08:16 5140875                    /usr/lib/x86_64-linux-gnu/libnuma.so.1.0.0
7fdca2f07000-7fdca3106000 ---p 0000a000 08:16 5140875                    /usr/lib/x86_64-linux-gnu/libnuma.so.1.0.0*** Aborted at 1522219134 (unix time) try "date -d @1522219134" if you are using GNU date ***
PC: @                0x0 (unknown)
*** SIGABRT (@0x7f78) received by PID 32632 (TID 0x7fdc97fff700) from PID 32632; stack trace: ***
    @     0x7fdcc767d390 (unknown)
    @     0x7fdcc1740428 gsignal
    @     0x7fdcc174202a abort
    @     0x7fdcc17827ea (unknown)
    @     0x7fdcc178b37a (unknown)
    @     0x7fdcc178f53c cfree
    @     0x7fdcbfee1f98 _ZNSt6vectorIN10statistics18StatisticsMapValueESaIS1_EE19_M_emplace_back_auxIJS1_EEEvDpOT_
    @     0x7fdcbfee1682 statistics::Statistics::GetHandle()
    @     0x7fdcbfee170c statistics::StatsCollectorImpl::StatsCollectorImpl()
    @     0x7fdcbc9a32fe landmark_triangulation::(anonymous namespace)::retriangulateLandmarksOfVertex()
    @     0x7fdcbc9a3d48 _ZNSt17_Function_handlerIFvRKSt6vectorImSaImEEEZN22landmark_triangulation12_GLOBAL__N_131retriangulateLandmarksOfMissionERKN6vi_map9MissionIdERKSt13unordered_mapIN5aslam7FrameIdEN5kindr7minimal26QuatTransformationTemplateIdEESt4hashISE_ESt8equal_toISE_EN5Eigen17aligned_allocatorISt4pairIKSE_SI_EEEEPNS8_5VIMapEEUlS4_E_E9_M_invokeERKSt9_Any_dataS4_
    @     0x7fdcc1da3c80 (unknown)
    @     0x7fdcc76736ba start_thread
    @     0x7fdcc18123dd clone
    @                0x0 (unknown)

Can you give me some advice ?
Thank you.

schneith, 2018-04-03:
I don't see a problem with the delayed start of the map builder in general; that should work. It might be a bit tricky to get the start/shutdown order of all threads right.

It seems that the statistics module causes the crash. If I remember correctly the module is turned off by default. Did you enable it? If so, could you try if the problem disappears when the module is off?

schneith, 2018-04-03:
I have added a test for the statistics module to test the multi-threaded access (https://github.com/ethz-asl/aslam_cv2/pull/22/files) and it did not fail on my machine. However, the access is very slow.

closed 2018-08-15 by schneith