ethz-asl/maplab #60: IDE

caoyifeng001, 2018-03-29:
which IDE do you using ?

mfehr, 2018-03-29:
[eclipse](https://github.com/ethz-asl/programming_guidelines/wiki/Catkin-Eclipse#adding-a-catkin-package-as-project-to-eclipse)

schneith, 2018-04-03:
This wiki entry might also be helpful: https://github.com/ethz-asl/maplab/wiki/Importing-maplab-to-Eclipse

plumewind, 2018-09-08:
@mfehr @schneith Hello, have you encountered this error? Thank you very much!

plumewind, 2018-09-10:
First of all, thank you very much for your reply. I have installed the 3.8 version of LLVM alone, but I still can't add plugins in elicpse. My elicpse version is as follows:

[image]

But it can't find this plugin:
[image]

Similarly, you can't add that plugin:
[image]

mfehr, 2018-09-14:
@plumewind if you are referring to building maplab, then please check the wiki on how to build it without IDE. The IDE or the LLVM plugin are not required to build, use or develop maplab.
That's why I can only refer you to my initial suggestion to check for tutorials online on how to install eclipse cdt and LLVM and try different versions.

plumewind, 2018-09-14:
@mfehr Thank you very much, I have successfully compiled maplab using catkin_build. However, based on the maplab framework, I want to develop some of my own things, so I want to create in the IDE, which is convenient. Also, the LLVM plugin for eclipse, found on the web page, they may not support the latest version,,,,,...

I thought that you would have a solution, so I always ask you, thank you very much for your help.In this case, then I will use the original way.Thank you very much for your help anyway.