ethz-asl/maplab #63: IMU timestamp not increasing strictly with Tango

Jeoker, 2018-04-07:
Hi all,
I am using Tango on Lenovo PB2-690Y, ROS Kinetic and Ubuntu 16.04.
When playing with ROVIOLI, I faced an issue that the maplab is complaining about timestamp not increasing strictly in IMU data. 
I found that the imu is only accurate to 1 ms, but it's sending data at 200 hz, which cause same timestamp in some data packages. so I go check the tango_ros_streamer wiki, but didn't find how to set the rate of imu.
Hence, I go to souce code of ROVIOLI and delete the assertion. It makes things work, but the result is terrible. I believe this is because the yaml files generated  by KALIBR have problems because of the same timestamp.
Do you guys know how to fix that? Many thanks in advance!

MetaI, 2018-04-09:
Dear @Jeoker ;
I had this same issue with Phab 2 Pro and after struggling with message headers sent from Tango Ros Streamer, Rosjava and eiffel library a bit finally installed ROS indigo along other actions in order to make it work. Please read issue #48 
I hope you can find it useful, otherwise let me know
Good luck!

Jeoker, 2018-04-09:
Hi Metal,
I actually read your post before. But switching everything back to Indigo may cost more than atteching an extra imu onto the device. So, for now we are trying to use an more accurate imu instead of the embeded one to see if it works. But still, thank you very much Metal, we will try your approach later if a new imu doesn't help!

closed 2018-08-15 by mfehr

MetaI, 2018-11-16:
Hey @Jeoker, did you finally solve it? I'm still having issues with imu measurements at 200Hz, and I was wondering about the right IMU rate at the time of calibrating. Now I wonder about the Yellowstone tablet and it's IMU rate. Perhaps the Yellowstone tablet from the paper was sending data at different frequency? 
Using the newer maplab march RC  
I'm receiving again errors with Lenovo Phab 2 Pro (~30Hz camera rate, ~200Hz imu rate):

 ```
[FilterBase::addUpdateMeas] Warning: included measurements at time 0.966738007 before safeTime 1

W0305 10:09:21.951208 32732 rovio-flow.cc:148] ROVIO rejected image measurement. Latency is too large.
F0305 10:09:22.006186 32732 imu-measurements-buffer-inl.h:11] Check failed: timestamp_nanoseconds > last_value.timestamp (1520240962222000000 vs. 1520240962222000000) Timestamps not strictly increasing.
```
Should I modify ROS Streamer in order to try lowering the IMU rate?

MetaI, 2018-11-19:
I was afraid of hearing that :S Jeoker, good that you advanced in your project and thanks for your answer! The requirements or mine are preciselly, a Tango device, so I'm hesitating about buying a Yellowstone tablet or trying to solve these issues. I tried to add the fixed ns delay computed by Kalibr in maplab July version, converted to ns, without luck yet

MetaI, 2018-11-19:
About the Tango Ros Streamer, I found that it is posible to adjust IMU measurement frequency by registering the imu listeners with fixed time as parameters instead of Android SDK constants, so maybe I try that before that before thinking in buying the Yellowstone tablet...
Though I would like to now if Yellowstone tablet would work work nowadays

Jeoker, 2018-11-19:
Hi Metal,
Have you tried to re-record a rosbag from the data you get from Tango? I was doing using maplab in a course project so I am not working it anymore. But I mean these may be something you would like to give a shot. It didn't work for me but there maybe some trick you can play with when re-recording a bag which I didn't know.

MetaI, 2019-01-03:
Hi Jeoker, I finally did it quite right lowering the imu rate to 50Hz and disabling health checking. Although not the ideal solution, I found that it estimates distances quite well. I will be testing Yellowstone tablet soon but for this device I thing I reached I good conclusion. Thank you very much your first post gave me a lot of clues of how to proceed. Take care!

Jeoker, 2019-01-03:
Awesome, I'm really happy to know that you have got it work! Happy new year!

gakkiHp, 2019-03-08:
@MetaI 
Hi, I want to know how to know imu rate and how to lower it？

cyitian, 2019-10-19:
@gakkiHp 
use
`rostopic hz /your_imu_topic `
 can know your imu rate.and use
` rosrun topic_tools throttle messages /old_imu_topic 50 /new_imu_topic`
can lower your imu rate ro 50Hz.
Finally,it really work!

MetaI, 2019-10-19:
Hi @gakkiHp,
I'm very sorry for the late repply, almost zombie reply, sorry. About the way that served to me, at first I tried the method that mentioned @cyitian, but then I found out that the values catched by the sensor class where not accurate and many times not syncing with the image. So I ended up modifiying the Android application in order to obtain values at a lower rate. More specifically, I modified the following class in the TangoRosStreamer app: 

[https://github.com/Intermodalics/tango_ros/blob/master/TangoRosStreamer/app/src/main/java/eu/intermodalics/tango_ros_streamer/nodes/ImuNode.java](url)

The most important change I did was around line 70: 

```
mSensorManager.registerListener(this, mRotationSensor, SensorManager.SENSOR_DELAY_FASTEST);
mSensorManager.registerListener(this, mGyroscopeSensor, SensorManager.SENSOR_DELAY_FASTEST);
mSensorManager.registerListener(this, mAccelerometerSensor, SensorManager.SENSOR_DELAY_FASTEST);
```

I changed the constant `SensorManager.SENSOR_DELAY_FASTEST` for a literal, for example `20000 `. Altering this value to a higher value than  `SensorManager.SENSOR_DELAY_FASTEST` will result in taking measurements at a lower rate or frequency, and, as I found later, with better accuracy. For me, 50Hz was the sweet spot. Of course, also you need to go through all the process of compiling the app successfully.  That was enough to track myself walking in the building using the embedded (and poor) IMU in the Lenovo Phab 2 Pro.

gakkiHp, 2019-10-21:
Thanks a lot! It helps a lot.

------------------ 原始邮件 ------------------
发件人: "MetaI"; 
发送时间: 2019年10月19日(星期六) 晚上8:06
收件人: "ethz-asl/maplab";
抄送: "gakkiHp"<1459351150@qq.com>;"Mention";
主题: Re: [ethz-asl/maplab] IMU timestamp not increasing strictly withTango (#63)

Hi @gakkiHp,
 I'm very sorry for the late repply, almost zombie reply, sorry. About the way that served to me, at first I tried the method that mentioned @cyitian, but then I found out that the values catched by the sensor class where not accurate and many times not syncing with the image. So I ended up modifiying the Android application in order to obtain values at a lower rate. More specifically, I modified the following class in the TangoRosStreamer app:
 
https://github.com/Intermodalics/tango_ros/blob/master/TangoRosStreamer/app/src/main/java/eu/intermodalics/tango_ros_streamer/nodes/ImuNode.java
 
The most important change I did was around line 70:
 mSensorManager.registerListener(this, mRotationSensor, SensorManager.SENSOR_DELAY_FASTEST); mSensorManager.registerListener(this, mGyroscopeSensor, SensorManager.SENSOR_DELAY_FASTEST); mSensorManager.registerListener(this, mAccelerometerSensor, SensorManager.SENSOR_DELAY_FASTEST);  
I changed the constant SensorManager.SENSOR_DELAY_FASTEST for a literal, for example 20000 . Altering this value to a higher value than  SensorManager.SENSOR_DELAY_FASTEST will result in taking measurements at a lower rate or frequency, and, as I found later, with better accuracy. For me, 50Hz was the sweet spot. Of course, also you need to go through all the process of compiling the app successfully.  That was enough to track myself walking in the building using the embedded (and poor) IMU in the Lenovo Phab 2 Pro.
 
&mdash;