ethz-asl/maplab #64: Where should I start  if I would like to add an extra sensor?

Doodleshr, 2018-04-11:
Hi guys,

I am trying to add an extra sensor to the existing system that is currently using stereo cameras and imu to provide another constraint to the optimization problem.
I am currently going through relevant modules and I would like to hear from you for some suggestions.

Thanks in advance for any help.
shr

dymczykm, 2018-04-11:
Hi, 
thanks for your question.

The following steps are necessary to integrate an additional sensor to the optimization problem:
* passing the data to the VIMap: you would either need to integrate it in the ROVIOLI message flow and pass it to the MapBuilder (https://github.com/ethz-asl/maplab/blob/master/applications/rovioli/include/rovioli/map-builder-flow.h) or load if offline after the map is created from your own sensor data format,
* storing the sensor information/calibration in the VIMap using SensorManager (https://github.com/ethz-asl/maplab/blob/master/map-structure/vi-map/include/vi-map/vi-map.h#L139),
* storing the sensor data in OptionalSensorData (https://github.com/ethz-asl/maplab/blob/master/map-structure/vi-map/include/vi-map/vi-map.h#L143),
* create the error terms for the sensor measurements (examples: https://github.com/ethz-asl/maplab/tree/master/algorithms/ceres-error-terms/include/ceres-error-terms),
* adding the error terms to the optimization, similar to the existing implementation, check the source file for details (https://github.com/ethz-asl/maplab/blob/master/algorithms/map-optimization/include/map-optimization/optimization-terms-addition.h).

If your additional sensor is a GPS sensor we have some of the implementation work done.

Doodleshr, 2018-04-11:
@dymczykm Thank you very much!!

closed 2018-04-11 by Doodleshr