ethz-asl/maplab #65: optv or optvi a single session map occurs error

YuTingLiu, 2018-05-02:
Thank you guys opensource this project.
I am following this page ([Preparing a single session map](https://github.com/ethz-asl/maplab/wiki/Preparing-a-single-session-map)) to optimize a single map that i save from ROVIOLI, and i have load VIO map to maplab. AND such as rtl, kfh and lc looks good. BUT when i want to test [Bundle adjustment](https://github.com/ethz-asl/maplab/wiki/Preparing-a-single-session-map#bundle-adjustment), error says:
maplab <test>:/$ optv 
I0502 17:50:21.639997 21785 optimization-problem.cc:175] VI gauge fixes: 
  num mission co-observability clusters: 1
  cluster 0:
    missions of cluster: [ 404b..0000 ]
    1st mission: 404b..0000
    1st vertex: 3cc7..0000
    position-fixed: 1
    rotation-fixed: kAll
    scale-fixed: 1
**E0502 17:50:21.693079 21785 solver.cc:487] Terminating: Can't use SPARSE_NORMAL_CHOLESKY with SUITESPARSE because SuiteSparse was not enabled when Ceres was built.**

Does anyone have this problem?

dymczykm, 2018-05-03:
Hi @YuTingLiu,
which system are you using? Did you build the ceres_catkin and [suitesparse](https://github.com/ethz-asl/suitesparse) packages?

YuTingLiu, 2018-05-04:
@dymczykm thanks, i am not sure this problem has been solved. what i have done is as follow:
## cite: http://ceres-solver.org/installation.html
* CMake
sudo apt-get install cmake
* google-glog + gflags
sudo apt-get install libgoogle-glog-dev
* BLAS & LAPACK
sudo apt-get install libatlas-base-dev
* Eigen3
sudo apt-get install libeigen3-dev
* SuiteSparse and CXSparse (optional)
* - If you want to build Ceres as a *static* library (the default)
*  you can use the SuiteSparse package in the main Ubuntu package
*  repository:
sudo apt-get install libsuitesparse-dev
## then, modify the CMakeList.txt:
-DSUITESPARSE_INCLUDE_DIR_HINTS=${suitesparse_INCLUDE_DIRS}
-DSUITESPARSE=ON

## catkin build, says:
-- Found LAPACK library: /usr/lib/liblapack.so;/usr/lib/libblas.so
-- Found BLAS library: /usr/lib/libblas.so
-- Found SuiteSparse 4.4.6, building with SuiteSparse.
-- Building without CXSparse.
-- Found Google Flags header in: /home/ll/maplab_ws/devel/include, in namespace: google
-- Found Google Log (glog). Assuming glog was built with gflags support as gflags was found. This will make gflags a public dependency of Ceres.
-- Building with OpenMP.
-- Found tr1/unordered_map/set in std::tr1 namespace.
-- Found shared_ptr in std::tr1 namespace using <tr1/memory> header.
-- Building Ceres as a shared library.
-- Do not build any example.

but when i return to maplab_console and try optv, error changed:

`
I0504 15:33:15.185940  6099 optimization-problem.cc:175] VI gauge fixes: 
  num mission co-observability clusters: 1
  cluster 0:
    missions of cluster: [ 7b95..0000 ]
    1st mission: 7b95..0000
    1st vertex: 57e3..0000
    position-fixed: 1
    rotation-fixed: kYaw
    scale-fixed: 0
*** stack smashing detected ***: /home/ll/maplab_ws/devel/lib/maplab_console/maplab_console terminated
*** Aborted at 1525419196 (unix time) try "date -d @1525419196" if you are using GNU date ***
PC: @     0x7fb981078428 gsignal
*** SIGABRT (@0x3e8000017d3) received by PID 6099 (TID 0x7fb9821a3b00) from PID 6099; stack trace: ***
    @     0x7fb9810784b0 (unknown)
    @     0x7fb981078428 gsignal
    @     0x7fb98107a02a abort
    @     0x7fb9810ba7ea (unknown)
    @     0x7fb98115c15c __fortify_fail
    @     0x7fb98115c100 __stack_chk_fail
    @     0x7fb963d3ba58 ceres::internal::(anonymous namespace)::OrderingForSparseNormalCholeskyUsingSuiteSparse()
    @     0x7fb963d3bc61 ceres::internal::ReorderProgramForSparseNormalCholesky()
    @     0x7fb963d5b75b ceres::internal::TrustRegionPreprocessor::Preprocess()
    @     0x7fb963d5308c ceres::Solver::Solve()
    @     0x7fb963d53fc9 ceres::Solve()
    @     0x7fb9643a4a8c map_optimization::solveWithOutlierRejection()
    @     0x7fb9643a9961 map_optimization::VIMapOptimizer::optimizeVisualInertial()
    @     0x7fb9643aa540 map_optimization::VIMapOptimizer::optimizeVisualInertial()
    @     0x7fb9645ce0e3 map_optimization_plugin::OptimizerPlugin::optimizeVisualInertial()
    @     0x7fb981df4ce4 common::CommandRegisterer::processCommand()
    @     0x7fb981df8c64 common::Console::RunCommand()
    @     0x7fb981dfb6b4 common::Console::RunCommandPrompt()
    @           0x4026b1 main
    @     0x7fb981063830 __libc_start_main
    @           0x402739 _start
已放弃 (核心已转储)`

closed 2018-05-04 by YuTingLiu