ethz-asl/maplab #67: Live visual inertial localisation and mapping using lenovophab2pro Tango phone

poudyalbot, 2018-06-02:
**TODO:** ADD YOUR QUESTION/PROBLEM DESCRIPTION HERE

---
Dear sir how can we setup live visual inertial odometry using Lenovo phab2pro Tango device since there is tutorial for how to setup using rosbag.
Thank you
Before submitting this issue, please consider the following **checklist**:

 - Consult the [FAQ](https://github.com/ethz-asl/maplab/wiki/FAQ) and [Known Issues](https://github.com/ethz-asl/maplab/wiki/Known-Issues) pages.
 - Make sure maplab was installed using the proper [installation instructions](https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu)
 - For bugs and build problems please add (if applicable):
   - Version of:
     - Operating System (32/64bit, 14.04/16.04/...)
     - Eigen
     - ROS
   - A backtrace (if maplab crashed)
   - Run `catkin config` in your workspace and post it here.
   - Details about the commands/scripts used.
   - Details about special parameter values or manual changes to the system.
   - Pictures/screen recordings
 - For general questions:
   - Please elaborate, we are no mind readers :)

mfehr, 2018-06-11:
Hi @poudyalbot 
If you intend to run maplab on the live sensor data being published by the [Tango ROS Streamer app](http://wiki.ros.org/tango_ros_streamer), this should work, at least in theory. However, be aware that we have not tested this. We have received some reports of people trying this, maybe you should have a look at these issues: https://github.com/ethz-asl/maplab/issues/48 https://github.com/ethz-asl/maplab/issues/63 

Regarding the question about how to do it: Download the Tango ROS Streamer app to the tango phone, establish a ros connection between your PC and the phone, stream the data and then use it as you would any other sensor. This mainly means calibrating the sensor and setting up the ros topics (using [Kalibr](https://github.com/ethz-asl/maplab/wiki/Initial-sensor-calibration-with-Kalibr) or try to convert the tango calibration to [our format](https://github.com/ethz-asl/maplab/wiki/Sensor-Calibration-Format))

poudyalbot, 2018-06-12:
Dear sir thank you

closed 2018-06-12 by poudyalbot