ethz-asl/maplab #70: Trajectory generate by ROVIOLI in Localization mode seems even worse than ROVIOLI in VIO mode

LeoKingBUAA, 2018-06-26:
Hi, all
I want to compare the trajectory generated by ROVIOLI in Localization mode and in VIO mode. So I do the following steps:
1. download dataset MH_04_difficult.bag
2. run ROVIOLI in VIO mode using MH_04_difficult.bag and save trajectory in EuRoc format.
3. use maplab_console Preparing a single-session map and generate a summary map for Localization mode.
4. run ROVIOLI in Localization mode and save trajectory in EuRoc format.
5. use evo to plot trajectory in VIO mode and in Localization mode. Following picture shows the result.
[image]
blue line is the trajectory in Localization mode 
green line is the trajectory in VIO mode
dotted line is groundtruth
it seems the trajectory even worse in Localization mode, so I want to ask what's the problem?   

**TODO:** ADD YOUR QUESTION/PROBLEM DESCRIPTION HERE

---

Before submitting this issue, please consider the following **checklist**:

 - Consult the [FAQ](https://github.com/ethz-asl/maplab/wiki/FAQ) and [Known Issues](https://github.com/ethz-asl/maplab/wiki/Known-Issues) pages.
 - Make sure maplab was installed using the proper [installation instructions](https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu)
 - For bugs and build problems please add (if applicable):
   - Version of:
     - Operating System (32/64bit, 14.04/16.04/...)
     - Eigen
     - ROS
   - A backtrace (if maplab crashed)
   - Run `catkin config` in your workspace and post it here.
   - Details about the commands/scripts used.
   - Details about special parameter values or manual changes to the system.
   - Pictures/screen recordings
 - For general questions:
   - Please elaborate, we are no mind readers :)

LeoKingBUAA, 2018-07-03:
Hi @dymczykm ,
I think I use the wrong localization data. I used vertices' position in VI-map as localization data before, but it's not the true localization results, actually. I found using flag --export_estimated_pose_to_csv can export the localization data in global coordinate. Comparing this data with the groundtruth the result seems better. 
[image]
To get the result,  I localize MH_04 dataset with the summary map which is generated by dataset MH_04. But in practical, it's impossible to use the same picture during mapping and localization. So, I generate a new summary map with datasets MH_01 to MH_03, and the localization result shows below.
[image]
The result seems not so good. I think the reason is little pose can be localized with map during the term, though the mapping environment and the localization environment are similar.
[image]
(red points are poses localized with summary map)
So, I want to ask why just little pose can be localize by map？ Is this an inherent problem of the feature point method? Or if there is a method to improve it ?

dymczykm, 2018-07-09:
@LeoKingBUAA 
Could you:
* provide here or link the the ``ncamera.yaml`` file you are using,
* when building the localization map, try out the commands I've linked above,
and report back? This would help diagnosing the issue.

JWDu, 2019-03-09:
Hi， @LeoKingBUAA 
I can't find the flag --export_estimated_pose_to_csv you mentioned，which command use the flag? I just ues the csv_export command to get the CSV data, and use the vertices' position in vertices.csv as the localization results. Is that wrong?
Looking forward to your reply！

amichayfeldman, 2020-06-21:
Hi,
I tried to execute Rovioli with localiztion like @LeoKingBUAA did - run MH1 with reference of summary of the same MH1 rosbag.

I got very bad results:
[image]
[image]

I run ROVIOLI as described on tutorials (create summary map and then run ROVIOLI with localization script) : `rosrun rovioli tutorial_euroc_localization /home/amichay/maplab_ws/save_folder_MH1_with_optimization_for_loc_localization MH1_calibration /home/amichay/maplab_ws/dataset/MH_01_easy.bag`

How can it be described?