ethz-asl/maplab #72: Image width check failed error

sabetAI, 2018-07-06:
I'm getting the following error:
F0706 12:48:41.574533  2842 visual-pipeline.cc:30] Check failed: input_camera_->imageWidth() == static_cast<size_t>(raw_image.cols) (640 vs. 1024).

I ran the following script:

```
#!/usr/bin/env bash
LOCALIZATION_MAP_OUTPUT=/home/gaspar/catkin_ws/src/maplab/applications/rovioli/maps
NCAMERA_CALIBRATION="$ROVIO_CONFIG_DIR/bumblebee2.yaml"
IMU_PARAMETERS_MAPLAB="$ROVIO_CONFIG_DIR/bb_imu-adis16488.yaml"
IMU_PARAMETERS_ROVIO="$ROVIO_CONFIG_DIR/bb_imu-sigmas-rovio.yaml"
REST=$@

rosrun

 rovioli rovioli \
  --alsologtostderr=1 \
  --v=2 \
  --ncamera_calibration=$NCAMERA_CALIBRATION  \
  --imu_parameters_maplab=$IMU_PARAMETERS_MAPLAB \
  --imu_parameters_rovio=$IMU_PARAMETERS_ROVIO \
  --datasource_type="rostopic" \
  --save_map_folder="$LOCALIZATION_MAP_OUTPUT" \
  --map_builder_save_image_as_resources=false \
  --optimize_map_to_localization_map=false $REST
```

OS: Ubuntu 16.04
ROS: Kinetic
---
Camera Calibration:

label: "Bumblebee2 - Jackal Calibration"
id: 412eab8e4058621f7036b5e765dfe812
cameras:
- camera:
    label: /front/left
    id: 54812562fa109c40fe90b29a59dd7798
    line-delay-nanoseconds: 0
    image_height: 480
    image_width: 640
    type: pinhole
    intrinsics:
      cols: 1
      rows: 4
      data: [788.4085367665094, 788.4085367665094, 512.5, 384.5]
    distortion:
      type: radial-tangential
      parameters:
        cols: 1
        rows: 4
        data: [0.0, 0.0, 0.0, 0.0]
  T_B_C:
    cols: 4
    rows: 4
    data: [0.0, 0.0, 1.0, 0.169,
           -1.0, 0.0, 0.0, 0.0,
           0.0, -1.0, 0.0, 0.217,
           0.0, 0.0, 0.0, 1.0]
- camera:
    label: /front/right
    id: 54812562fa109c40fe90b29a59dd7723
    line-delay-nanoseconds: 0
    image_height: 480
    image_width: 640
    type: pinhole

    intrinsics:
      cols: 1
      rows: 4
      data: [788.4085367665094, 788.4085367665094, 512.5, 384.5]
    distortion:
      type: radial-tangential
      parameters:
        cols: 1
        rows: 4
        data: [0.0, 0.0, 0.0, 0.0]
  T_B_C:
    cols: 4
    rows: 4
    data: [0.0, 0.0, 1.0, 0.289,
           -1.0, 0.0, 0.0, 0.0,
           0.0, -1.0, 0.0, 0.217,
           0.0, 0.0, 0.0, 1.0]

dymczykm, 2018-07-07:
@sabetAI it seems that you calibration says your camera has a resolution of 640x480:
```
image_height: 480
image_width: 640
```
while it actually has a larger resolution of 1024x768. If you correct this inconsistency, it should work.

sabetAI, 2018-07-11:
It worked, thank you!

closed 2018-07-11 by sabetAI