ethz-asl/maplab #75: Where should I start if I would like to start a ros node for surface reconstruction using voxblox？

plumewind, 2018-07-20:
Hi guys,

I want to create a TSDF grid map using the maplab framework, and I have noticed that this function can be implemented with console commands —— create_tsdf_from_depth_resource. However, I don't know how to visualize the map  throught voxblox in Rviz, and I feel that it is impossible  in maplab.
In order to create a TSDF grid map, I want to know how to create a node with the same function as the command create_tsdf_from_depth_resource.

Thanks in advance for any help.

mfehr, 2018-07-20:
Hi @plumewind 

You are right, there is currently no command to visualize the reconstruction in rviz directly. You need to open the PLY file. However this should be quite easy to implement and a welcome addition to maplab. Basically all you need to do is porting some of the mesh publication logic from [here](https://github.com/ethz-asl/voxblox/blob/master/voxblox_ros/src/tsdf_server.cc#L343) to [here](https://github.com/ethz-asl/maplab/blob/master/console-plugins/dense-reconstruction-plugin/src/dense-reconstruction-plugin.cc#L356) and/or [here](https://github.com/ethz-asl/maplab/blob/master/console-plugins/dense-reconstruction-plugin/src/dense-reconstruction-plugin.cc#L400) or into a new command.

Could you ellaborate what this rosnode should do? Do you want a full end-to-end pipeline, i.e. it loads a rosbag and outputs a voxblox reconstruction to RVIZ based on the optimized  pose graph?

plumewind, 2018-07-25:
Hi @mfehr 

First of all, thank you very much for your reply. Now, I know that maplab's map reconstruction process can't be displayed on  RVIZ, so I'm constantly looking for ways. However, because my academic is still shallow, and should the PLY file be converted to the voxblox message format for display? So some questions still don't know how to deal with it.

About that ROS node, its function is to realize the map reconstruction function of maplab, and can be displayed in RVIZ, and the meaning of the rovioli node is the same. Moreover, I wanted to do this work from the beginning and found that the display is a big problem.

Thank you very much for your help.

mfehr, 2018-07-29:
I edited my answer above, because it was a bit misleading. 

The pointers I gave you are about directly visualizing a voxblox TSDF map by creating a surface mesh and converting and publishing it as custom voxblox ROS message (which rviz can visualize). Meaning if you add this logic to the reconstruction command you can directly visualize the reconstructions without writing a PLY file.

Regarding the ros node you are trying to build. If all you want is to automate the process to get from rosbag to reconstruction, the easiest way I can think of is adding the sequence of operations needed (triangulation, relax, loop close, optimize, stereo matching, tsdf integration, surface reconstruction, visualize mesh) to the shutdown section of rovioli, where the map is usually processed into an optimized localization map ([here](https://github.com/ethz-asl/maplab/blob/master/applications/rovioli/src/map-builder-flow.cc#L90))

plumewind, 2018-08-15:
@mfehr Thank you very much for your help, I am trying, thank you.

closed 2018-09-19 by plumewind