ethz-asl/maplab #77: Point cloud in VIO mode

zhukovv, 2018-07-25:
Is there a way to get a pointcloud from VI-map online?
I have `--rovioli_visualize_map` enabled, but it publishes a cloud (landmarks) just in the end of the session. I guess because initializeLandmarks is called just in the saveMapAndOptionallyOptimize function.
Can I get 3D points in some form during a session - I want to do plane segmentation based on the data.

Thank you!

mfehr, 2018-07-25:
Hi @zhukovv 
As you already pointed out, the landmarks are only initialized (and triangulated) during the shutdown of the estimator.
However, @eggerk implemented something similar for some other experiment. The experiment itself was not released, but some of the new tools and algorithms are already in the codebase. Check out `algorithms/online_map_builders/`, there is a `KeyframedMapBuilder` in there that keyframes the VIMap while building it (duh) and also initializes the landmarks incrementally. It looks like the keyframing can be disabled. This code is currently untested and was used in another context, so all bets are off. You might be able to use it though with some adaptations to the interface. It would be cool if you could let us know how this turns out :)

zhukovv, 2018-07-25:
Hi @mfehr 
Sounds promising, thank you very much!
I'll get back when I have some results :)

drNoob13, 2019-06-19:
Hello @zhukovv,

Were you able to make it work?

Thanks!