ethz-asl/maplab #78: Run other dataset with ROVIO

wendaoyuchen, 2018-07-26:
Hi, I want to test maplab with the dataset which I recorded by ROS with ZR300.
And I write three scripts(tutorial_realsense, tutorial_realsense_live,tutorial_realsense_localization_live) in "/home/***/maplab_ws/src/maplab/applications/rovioli/scripts/tutorials",  three YAML files(imu-zr300-maplab.yaml, imu-zr300-maplab.yaml, ncamera-zr300.yaml) in "/home/liuzhen/maplab_ws/src/maplab/applications/rovioli/share".
So what should I do to run these scripts 
`run rovioli tutorial_realsense save_folder datasets.bag`
like 
`run rovioli tutorial_euroc save_folder MH_01_easy.bag `?

dymczykm, 2018-08-07:
@wendaoyuchen If your bag contains the correct data + the topic names are correctly configured in your yaml files (please check) you can still use a script identical to ``rosrun rovioli run_rovioli``, just make sure it points to the correct yaml files, here and below:
https://github.com/ethz-asl/maplab/blob/master/applications/rovioli/scripts/run_rovioli#L10

wendaoyuchen, 2018-08-10:
Thanks for your responds! @dymczykm 
I  has solved this problem . I should change the file authority like below:
chmod +x tutorial_realsense
and it works. Beside, if some other guys trouble in this problem. You should vim or gedit a new file named _tutorial_realsense_ , and edit it rather than copy the file  _tutorial_euroc_  and rename it.

closed 2018-08-15 by mfehr