ethz-asl/maplab #80: Online map generation and loop closure for usage in ROVIOLI Localization mode

TSC21, 2018-07-29:
Dears,
I didn't find the answer for this on the Wiki (probably I didn't search in depth), but is it possible to run ROVIOLI in Localization mode from a map being online reconstructed and optimized (using loop closure)? I ask this because from the references, I only see that you use a predefined optimized map that is loaded and then ROVIOLI localizes itself against it.
Thank in advance.

TSC21, 2018-07-29:
Is https://github.com/ethz-asl/maplab/tree/master/algorithms/online_map_builders something to look at, regarding my own question?

mfehr, 2018-07-29:
hi @TSC21 
You're right, this feature is currently not available. A map is being built online (stream map builder), but its landmarks are neither initialized nor triangulated and no loop closure or optimization is performed. Therefore it currently cannot be used by the localizer online. The keyframed map builder is at least triangulating the landmarks and was used in some unreleased online mapping experiments. It's a good starting point though and most of the components needed to get some basic form of online localization running in ROVIOLI are available in the code.

TSC21, 2018-07-29:
@mfehr thanks for your feedback. Is this something you are interested to support in further Maplab/ROVIOLI iterations?

mfehr, 2018-07-29:
we are interested and are looking into it, however we can't say or promise anything yet in terms of if or when this will be available.

TSC21, 2018-07-29:
Is that effort supportable? Would be interested on helping if that's possible and you are interested as well.

mfehr, 2018-08-02:
Could you contact us by email?

TSC21, 2018-08-02:
Yes. To which e-mail?

K0stIa, 2018-10-18:
Hi, @mfehr.

I am interested in the same question as @TSC21. Is there any progress in maplab about it?

mfehr, 2018-10-18:
[image]

TSC21, 2018-10-19:
@K0stia it was on purpose 😊 it's a gif to basically say that this is an ongoing work.

K0stIa, 2018-10-19:
Got it. 
p.s. I hope you were laughing after my second post!

Cheers, -K,

mhkabir, 2018-10-31:
@mfehr Interested in taking this up and creating a baseline working solution. Would you have to discuss over email?

K0stIa, 2018-10-31:
@mhkabir I am interested in the same thing too.

bohell, 2020-07-23:
any updates about this? guys @mfehr

MigVega, 2022-12-17:
Hi all, I am also very interested in something like that! Is there something already implemented in Matlab 2.0 which enables running on localization mode and creating at the same time an updated aligned map? Maybe also with LiDAR sensors?
I would be interested to contribute to that solution @mfehr !