ethz-asl/maplab #83: Transform other type map to maplab  map type

QinZiwen, 2018-08-01:
I have other type map(for example get from photoscan), How transform this map for which maplab console can load? I don't how to do this, thanks.

mfehr, 2018-08-02:
Hi @QinZiwen 

I'm not familiar with the map format of photoscan and what it contains exactly. Since it's a photogrammetry pipeline, at the very least camera poses and calibrations. I can only give you some pointers to places in the code where we build a map from some data source (or export a map to another format), you will have to come up with a way to read the data from photoscan maps yourself.

 * Here we export our map format to CSV: [link](https://github.com/ethz-asl/maplab/tree/master/tools/csv-export)
 * Here we export our map to the bundler format to be digested by PMVS: [link](https://github.com/ethz-asl/maplab/tree/master/tools/csv-export)
 * This is a map generator for unit testing, so it will contain all the examples you need to fill a map with data: [link](https://github.com/ethz-asl/maplab/blob/master/map-structure/vi-map/include/vi-map/test/vi-map-generator.h)

QinZiwen, 2018-08-02:
Thanks @mfehr

closed 2018-08-02 by mfehr