ethz-asl/maplab #84: memory leak in loopclosure

democheng, 2018-08-01:
I find one bug(memory leak) in loopclosure:
algorithm/loopclosure/matching-based-loopclosure/include/matching-based-engine.h
line 26, virtual ~MatchingBasedLoopDetector() = default;
This Destructor does nothing, so database_, descriptor_index_to_keypoint_id_ and index_interface_ won't be cleared.

I rewrite it:
virtual ~MatchingBasedLoopDetector() {
      Clear();
}

dymczykm, 2018-08-02:
@democheng I don't think you're right. All of the members you mentioned are either shared pointers (smart pointers in general) or just STL containers (some of which contain smart pointers), they will get properly destructed when the ``MatchingBasedLoopDetector`` object is destructed.

Do you have any signs of a memory leak?

democheng, 2018-08-03:
@dymczykm Yeah, for C++, you are totally right.
I'm sorry for the missing information:
I use the _MatchingBasedLoopDetector_ on IOS, and complie it by xcode(v9.4, LLVM).
I use the xcode->Product->Profile>leak to analyze the usage of memory, and there is one leak because of _MatchingBasedLoopDetector_, here is the screenshot.
I guess the leak is caused by the difference of LLVM and GCC.
[image]

closed 2018-10-26 by democheng